compas-dev/compas #12: intersection_lines_xy has confusing input and output

brgcode, 2017-11-09:
it is unclear if the function only returns an intersection point if all lines intersect at one point, or all intersection points between pairs of lines in the set.

brgcode, 2017-12-05:
the current implementation has been replaced by a NotImplementedError...

brgcode, 2017-12-08:
it should return one approximate intersection point

Rippmann, 2017-12-10:
I would suggest to return all points of all possible intersections. I would leave it to the user to then compute the barycenter etc. Not sure how useful such a function is... especially if you give it more than three lines. You quickly get intersection points far away from an approximate intersection.

brgcode, 2018-06-25:
okay the function has been removed.
there is a gist with an implementation based on numpy for finding the "best-fit" intersection of four non-concurrent lines. 

https://gist.github.com/brgcode/0fc54568ca2bf51044a2a9d52db8fcb3

closed 2018-06-25 by brgcode