compas-dev/compas #24: Missing check in edge_coordinates

robin-oval, 2017-12-20:
Missing check in edge_coordinates in datastructures._mixins.geoemtry: returns the coordinates of two vertices, even if they are not connected by an edge.

brgcode, 2018-04-18:
hi robin,

the check is left out on purpose. it is not the responsibility of this function to check if there exists a connection between the two vertices, and in most use cases (for example in a loop over all edges) it is entirely redundant.

however, i can see that the name is misleading. perhaps the function should just be removed because this task can be accomplished in many different other ways.

where/when did you have this problem?

best,
tom

robin-oval, 2018-04-18:
Hi Tom,

It actually makes sense to leave the check out of that function. Not sure
when it occurred, maybe just when I was applying some rules on an invalid
input. Just wasn't sure if it was on purpose.

Best,
Robin

closed 2018-04-18 by brgcode