compas-dev/compas #29: mesh.from_lines   boundary face does not get deleted

Rippmann, 2018-01-12:
Strange behaviour (occurred on larger input geometries). Sometimes the boundary face does not get deleted with delete_boundary_face=True. However, another face get deleted instead. 

        if delete_boundary_face:
            mesh.delete_face(0)

It seems the the boundary face is not always indexed 0.

brgcode, 2018-02-01:
there are a few things to consider here.

the `from_lines` class method of `Mesh` first makes a network and then finds its faces. the first face it finds is the face containing the most *bottom-left* vertex, which usually is the outside face if the mesh is connected. it then breaks this face into smaller parts if it contains leaves of the network of lines. in this case, the *outside* face has effectively already been deleted, because it then has been replaced by the broken up smaller faces.

however, the above logic does not apply if there are no leaves in the *first* face. then this face is the *outside* face and could be deleted.

the vertices and found faces are then used to construct a `Mesh` using the class method `from_vertices_and_faces`. in the previous implementation, when requested, the outside face was deleted on this mesh, still assuming that this would be the first face of the mesh. this assumption is not necessarily true since the faces are not always added in the order they were found by the face finding algorithm.

the below examples (hopefully) illustrate what i tried to describe :)

```python
import compas
from compas.files import OBJ

obj = OBJ(compas.get('lines.obj'))

lines = [(obj.parser.vertices[u], obj.parser.vertices[v]) for u, v in obj.parser.lines]
mesh = Mesh.from_lines(lines, delete_boundary_face=False)

print(mesh.number_of_faces())  

# this will print 25
# which is correct because the face that was first found is indeed on the boundary
# but it contains leaves and will therefore be replaced by smaller, broken up faces
# note that setting the delete flag to `True` will result in an error
# because the algorithm will attempt the face with key `0`, which no longer exists
```

```python
import compas
from compas.files import OBJ

obj = OBJ(compas.get('lines_noleaves.obj'))

lines = [(obj.parser.vertices[u], obj.parser.vertices[v]) for u, v in obj.parser.lines]
mesh = Mesh.from_lines(lines, delete_boundary_face=False)

print(mesh.number_of_faces())  

# this will print 26, which is correct because the boundary face was not broken into smaller pieces
# because the set of lines has no leaves, and was thus not implicitly deleted
```

```python
import compas
from compas.files import OBJ

obj = OBJ(compas.get('lines_noleaves.obj'))

lines = [(obj.parser.vertices[u], obj.parser.vertices[v]) for u, v in obj.parser.lines]
mesh = Mesh.from_lines(lines, delete_boundary_face=True)

print(mesh.number_of_faces())  

# this will print 25
# which is correct because the boundary face was the first face and has been explicitly deleted
```

note that the current implementation (as made available through hotfix `0.1.0-#29`) may still yield unexpected results, but it at least works as originally intended...

brgcode, 2018-02-01:
the above explanation is correct, but there still seems to be an issue...

brgcode, 2018-02-01:
the issue is/was that in cases where the input geometry is 100% symmetrical around the *bottom-left* vertex, for example if the input is an orthogonal grid, the *find faces* algorithm starts with one of the internal faces rather than the *outside* face.

this should be solved now...

closed 2018-02-01 by brgcode