compas-dev/compas #51: Object names in Rhino 6 do not accept braces or brackets!

Rippmann, 2018-02-27:
Something to be aware of. Potential problems when dumping dictionary and list objects using, for example:
        input = json.dumps({'type': 'f','key': fkey, 'data': mesh.facedata[fkey], 'vertices': mesh.face[fkey]})
        rs.ObjectName(poly, input)

Is there already something in compas_rhino that depends on braces or brackets in object names?

See:
https://mcneel.myjetbrains.com/youtrack/issue/RH-44359

and here:
https://discourse.mcneel.com/t/rs-objectname-doesnt-work-with-strings-containing-curly-braces-in-rhino-6/55942/2

brgcode, 2018-03-14:
there is nothing in compas relying on this.

however, typically, when converting data structures to geometry we tend to write the attributes of the vertices, edges and faces in the name of the corresponding Rhino objects as json-serialised dictionaries.

these names then contain braces.

it is very convenient to be able to do so, because it makes the reverse process of (re)creating a data structure from rhino geometry very straightforward.

brgcode, 2018-04-18:
by the way, as far as i can tell, braces are allowed in the name, just not at the beginning...

closed 2018-08-21 by brgcode