compas-dev/compas #56: compas_rhino.helpers.artist.mixins.faceartist - draw_facelabels does not work for volmesh

juney-lee, 2018-03-14:
Line 116 of the draw_facelabels function in compas_rhino.helpers.artist.mixins.faceartist:

```
'pos'  : self.datastructure.face_center(key)
```

volmesh currently has no built-in function called face_center, and therefore the faceartist cannot draw a label on its (half)faces. One line function that should be added in the future...

Using "face" in a volmesh context is a bit confusing, since "face" here is actually just drawing one of the two conincident halffaces with the smaller key... Should the "face_name" then in this case include both halfface keys, similar to how edgelabels are displayed? For example, if the two halffaces 0 and 1 are coincident, faceartist will draw halfface 0; the label here could be "0-1." Or maybe it's okay with the way it is, since faceartist will be used almost always for visualization/selecting purposes only.

brgcode, 2019-06-16:
is this still the case since the volmesh pull request

juney-lee, 2019-06-17:
Before, there was only ``volmesh.halfface_center``, which is why faceartist didn't work. 

Now there is also ``volmesh.face_center``, which does the same thing as ``volmesh.halfface_center``.

closed 2019-06-17 by juney-lee