compas-dev/compas #73: Cannot import `compas_rhino` from Grasshopper (GHPython)

gonzalocasas, 2018-03-20:
Currently it is not possible to import any `compas_rhino` functionality from Grasshopper, mainly because of three lines:
 - [this one on `layers.py`](https://github.com/compas-dev/compas/blob/master/src/compas_rhino/utilities/layers.py#L10). 
 - [this one on `objects.py`](https://github.com/compas-dev/compas/blob/master/src/compas_rhino/utilities/objects.py#L8). 
 - [and this one on `drawing.py`](https://github.com/compas-dev/compas/blob/master/src/compas_rhino/utilities/drawing.py#L37)

Regarding the decision of an additional `compas_grasshopper` package, the impossibility to import the namespace is very confusing, especially because the error message is also very misleading.

To reproduce, simply drop a GHPython component on an empty GH canvas, and type `import compas_rhino`:

[image]

Rippmann, 2018-03-28:
I ran into the same problem today.

gonzalocasas, 2018-03-28:
Pull request available: https://github.com/compas-dev/compas/pull/74

brgcode, 2018-05-09:
solved in pull request #74

commit 887d05a3 (2018-04-30): fix #73: try..except all layer-related imports that fail in GHPython context
commit ef2caeb4 (2018-04-30): Merge pull request #74 from compas-dev/fix/compas_rhino_import_from_gh

fix #73: try..except all layer-related imports that fail in GHPython context
closed 2018-05-09 by brgcode