compas-dev/compas #76: Bug in mesh.copy()

romanarust, 2018-03-26:
Hi all,

There is TypeError when calling Mesh.copy():
   
    import compas
    from compas.datastructures import Mesh

    mesh = Mesh.from_obj(compas.get('mesh.obj'))
    mesh_cp = mesh.copy()

File "C:\Users\rustr\workspace\compas\src\compas\datastructures\mesh\mesh.py", line 612, in _clean_vertices
    if vertices[0] == vertices[-1]:
TypeError: 'map' object is not subscriptable

brgcode, 2018-04-18:
hi romana,

i can't reproduce the error. if i run the snippet above it works just fine.
can you give me a bit more info?

where are you running this (Sublime, Rhino, GH, ...)?
which version of Python?
which platform?

best,
tom

romanarust, 2018-04-18:
Hi Tom,
I realized that is a python issue: It works fine with python 2.7, but the error occurs with python 3.6.
windows, python 3.6
Best, romana

brgcode, 2018-04-18:
ha okay, the map function returns a map object in Python 3.x...

solved with hotfix 0.1.0-#76

closed 2018-04-18 by brgcode