compas-dev/compas #80: mesh_cull_duplicate_vertices

Rippmann, 2018-04-05:
to be changed to keep face attributes during sanatize step. E.g. like this:

    # sanitize affected faces
    new_faces = {}
    for fkey, **attr in mesh.faces(True):**
        face = []
        seen = set()
        for key in mesh.face_vertices(fkey):
            if key in keys_pointer:
                pointer = keys_pointer[key]
                if pointer not in seen:
                    face.append(pointer)
                    seen.add(pointer)
            else:
                face.append(key)
        if seen:
            new_faces[fkey] = face, **attr**

    for fkey in new_faces:
        del mesh.face[fkey]
        mesh.add_face(new_faces[fkey][0], fkey, **new_faces[fkey][1])**

Rippmann, 2018-06-26:
I just had problems with the resulting mesh from mesh_cull_duplicate_vertices. It seems that the halfedges of the previous mesh are not properly updated etc. E.g. if I use vertices_on_boundary() on the resulting mesh, I get wired results (boundary vertices at internal vertices).

Rippmann, 2018-06-26:
might be as simple as replacing         
'del mesh.face[fkey] '
with 
'mesh.delete_face(fkey)'
in the last loop.

brgcode, 2018-08-20:
i can't reproduce this at all. is it possible that this is in reference to old code? current implementation is as follows (method of the mesh class):

```python
    def cull_vertices(self):
        """Remove all unused vertices from the mesh object.
        """
        for u in list(self.vertices()):
            if u not in self.halfedge:
                del self.vertex[u]
            else:
                if not self.halfedge[u]:
                    del self.vertex[u]
                    del self.halfedge[u]
```

brgcode, 2018-08-20:
just for my understanding, what is this function actually supposed to do?
merge vertices that are geometrically equivalent (up to the given precision) to the same point?

Rippmann, 2018-08-20:
For example:
Imagine you have two meshes from a procedure like creating two delaunay meshes from two patches sharing the same edge (constraint delaunay). Since you started with the same number of division points on the shared edge, you know that the two meshes align perfectly at those vertices. This function should help to join these two meshes into one.

brgcode, 2018-08-20:
shall we then call it join_meshes with an option to cull the duplicates

Rippmann, 2018-08-20:
sure. sounds good!

Rippmann, 2018-09-20:
solved with Tom's new function

closed 2018-09-20 by Rippmann