compas-dev/compas #83: Inconsistent assumptions about vector space (dimension)

gonzalocasas, 2018-04-08:
The vector operations on `compas.geometry.basic` ([source](https://github.com/compas-dev/compas/blob/master/src/compas/geometry/basic.py)) are inconsistent in the way they assume the vector space. Some of them work on arbitrary length vectors, while others are hard-coded to 3D vectors, and documentation only refers to support of 3D vectors, even in the cases where the function support arbitrary length.

For example, `dot_vectors`, `add_vectors`, `subtract_vectors`, `scale_vector` and many others work on any length, but `cross_vectors`, `length_vector`, `normalize_vector` and others are hard-coded to 3D.

gonzalocasas, 2018-04-08:
/cc @stefanaparascho

brgcode, 2018-04-18:
okay, but are you asking for general support for arbitrary length vectors? or just for consistency?

gonzalocasas, 2018-04-18:
Consistent 3D vector support would be preferable, because a few of the functions can be sped up significantly based on that assumption. I don't think there's much use for arbitrary length vectors.

brgcode, 2018-08-20:
perhaps we should just remove

- `norm_vector(s)`
- `power_vector(s)`

as these are functions that make more sense in a general vector calculus context and not so much in a geometrical one.

closed 2020-04-14 by brgcode