compas-dev/compas #86: vtkvoxels: data to color mapping assumes positive numbers only

worbit, 2018-04-27:
First of all congrats, Andrew. Great work on the vtk viewers!
I propose to add one line in `draw()`:
`u = self.data`
`u -= min(u)`
`u /= max(u)`
`u *= 255`
in line 109, and accordingly `from numpy import max,min` in line 27.
With this, the entire range of `data` gets scaled to the HSB spectrum. This is essential vor signed distance fields as inside values are negative. Can also discuss alternative color schemes, e.g. one palette for positives, one for negatives, specific color white for zero...

andrewtliew, 2018-04-29:
Thanks! Great idea, just made a change where:

0-128 is blue to green, 128-255 green to red.
The midpoint is fixed green, so positive scalars are now always "warm" colours, and negative (if any) are always "cool". The colours are scaled by the absolute largest scalar.

The opacity slider now works from 128 outwards in both directions, i.e it hides both positive and negative from smaller values first.

worbit, 2018-05-02:
thanks andrew, this is a great improvement. i have, if i may, yet an additional suggestion: green is known to be (not very scientific formulation, i know, but i can't find the article a.t.m.) the hue where the human eye has most difficulties to discern changes. one suggestion could be to either use (or give the possibility to use) a log scale or maybe try to access the matplotlib colormaps: [matplotlib.org/users/colormaps.html](https://matplotlib.org/users/colormaps.html)

andrewtliew, 2018-05-02:
A log scale will definately be included as I have used the voxels wth the FE package and found that stress concentrations affect having a good colour range. I will make it as a slider so they can adjust the power of the new scaling.

One idea could be to allow the colormap to be added as an argument to VtkVoxels, and/or as you say, tap already into matplotlib colormaps by name.

brgcode, 2020-04-14:
is this still relevant?

andrewliew, 2020-04-15:
I think this is ok to close.

closed 2020-04-15 by andrewliew