compas-dev/compas #91: compas.verify() AttributeError

Achillx, 2018-06-05:
**ISSUE**
Running compas.verify() returns the following error:
```
>>> compas.verify()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/axydis/PROJECTS/compas-dev/compas/src/compas/__init__.py", line 146, in verify
    current = installed()
  File "/Users/axydis/PROJECTS/compas-dev/compas/src/compas/__init__.py", line 183, in installed
    installed_packages = pip.get_installed_distributions()
AttributeError: 'module' object has no attribute 'get_installed_distributions'
```
**REASON**
The `installed()` function uses the `get_installed_distributions()` from pip which, as per this thread [https://github.com/pypa/pip/issues/5243](https://github.com/pypa/pip/issues/5243), has been moved into pip._internal to signify that its is not supported and may change anytime without notice. It is strongly discouraged to use it.

**PROPOSED FIX**
```
def installed():
    import pkg_resources
    installed_packages = pkg_resources.working_set
    flat_installed_packages = [package.project_name for package in installed_packages]
    return sorted(flat_installed_packages, key=str.lower)
```

gonzalocasas, 2018-06-05:
I also bumped into it and fixed this (with the exact same proposed fix) on this pull request: https://github.com/compas-dev/compas/pull/89  🙂 

@brgcode one more reason to merge? ;)

brgcode, 2018-06-06:
solved on develop branch with pull request #89

closed 2018-06-06 by brgcode