compas-dev/compas #92: mesh_from_polylines / mesh_from_face_coordinates

Rippmann, 2018-06-06:
create a mesh from polylines/polygons
polyline/polygon as list of points

name? mesh_from_polylines / mesh_from_polygons / mesh_from_face_coordinates  ??

as a starting point a snippet working in Rhino:

import Rhino
import rhinoscriptsyntax as rs
import scriptcontext as sc
tempMesh = Rhino.Geometry.Mesh()

from compas.datastructures.mesh import Mesh
from compas_rhino.helpers.artists.meshartist import MeshArtist

from compas.utilities import geometric_key

#pts = rs.GetPoints("Select Points")

polylines = rs.GetObjects("Select Polylines",4)

vertices_geo = {}
faces_geo = []
for poly in polylines:
    curve = rs.coercecurve(poly)
    rc, polyline = curve.TryGetPolyline()
    pts = [pt for pt in polyline][:-1]
    face = []
    for pt in pts:
        geo_key = geometric_key(pt)
        face.append(geo_key)
        vertices_geo[geo_key] = pt
    faces_geo.append(face)
    
map_key = {}
vertices = []
for index, (key, value) in enumerate(vertices_geo.items()):
    map_key[key] = index
    vertices.append(value)

faces = []
for face_geo in faces_geo:
    face = [map_key[key] for key in face_geo]
    faces.append(face)

mesh = Mesh.from_vertices_and_faces(vertices, faces)

brgcode, 2018-06-18:
i think the name should be `mesh_from_polygons`. even though there is the slight implication that polygons are planar figures. depends who you ask... 

polylines are just not necessarily closed and may self-intersect.

brgcode, 2018-06-20:
added on the develop branch

closed 2018-06-20 by brgcode