compas-dev/compas #93: Rhino 6 import error:   import OBJ from compas.files & compas.files import PLYreader

Rippmann, 2018-06-06:
Message: cannot import OBJ from compas.files

same for from compas.files import PLYreader

Traceback:
  line 8, in <module>, "C:\Users\rippmanm\Documents\GitHub\compas\src\compas\datastructures\mesh\mesh.py"
  line 1, in <module>, "C:\Users\rippmanm\Documents\GitHub\compas\src\compas\datastructures\mesh\__init__.py"
  line 11, in <module>, "C:\Users\rippmanm\Desktop\io.py"
  line 32, in <module>, "C:\Program Files\Rhino 6\Plug-ins\IronPython\Lib\tempfile.py"
  line 6, in <module>, "C:\Program Files\Rhino 6\Plug-ins\IronPython\Lib\mimetools.py"
  line 80, in <module>, "C:\Program Files\Rhino 6\Plug-ins\IronPython\Lib\httplib.py"
  line 94, in <module>, "C:\Program Files\Rhino 6\Plug-ins\IronPython\Lib\urllib2.py"
  line 8, in <module>, "C:\Users\rippmanm\Documents\GitHub\compas\src\compas\files\obj.py"
  line 73, in <module>, "C:\Users\rippmanm\Documents\GitHub\compas\src\compas\files\__init__.py"
  line 10, in <module>, "C:\Users\rippmanm\Documents\GitHub\compas\src\compas\datastructures\network\network.py"
  line 1, in <module>, "C:\Users\rippmanm\Documents\GitHub\compas\src\compas\datastructures\network\__init__.py"
  line 86, in <module>, "C:\Users\rippmanm\Documents\GitHub\compas\src\compas\datastructures\__init__.py"
  line 11, in <module>, "C:\Users\rippmanm\Desktop\io.py"

gonzalocasas, 2018-06-07:
There is an update available for Rhino 6, which apparently fixes the issue (for some reason that I don't understand), but I was able to reproduce before the update and it started working right after it.

brgcode, 2018-06-18:
can we close this?

closed 2018-06-20 by brgcode