compas-dev/compas #94: Documentation Example typo: Network Equilibrium

tkmmark, 2018-06-11:
https://compas-dev.github.io/main/examples/fofin/00-cablenet.html

change (a) to (b)

(a)
for key, attr in network.vertices(True):
    index = key_index[key]
    attr['x'] = xyz[index, 0]
    attr['y'] = xyz[index, 1]
    attr['z'] = xyz[index, 2]

(b)
for key, attr in network.vertices(True):
    index = key_index[key]
    attr['x'] = xyz[index][0]
    attr['y'] = xyz[index][1]
    attr['z'] = xyz[index][2]

brgcode, 2018-06-12:
hi makr,

that doesn't seem to make sense, since `fd_numpy` returns NumPy arrays.
perhaps we should change that though...

best,
tom

tkmmark, 2018-06-12:
Yes, sorry.  I was looking at the wrong page; was actually doing an XFunc example and the syntax for that is correct, which returns a list.

tkmmark, 2018-06-12:
Also, I am noticing a few little things when I am playing with Compas, and I am not sure if that is a syntax change from the earlier version to the newer version of Python. For instance:

(1) TNA had a bunch of Print command that is missing the bracket
(2) The basic geometry example, for example Point, needs the square bracket removed:
p1 = Point([1, 2, 3]) -> p1 = Point (1,2,3)