compas-dev/compas #96: convex_hull_xy

salmamzfr, 2018-06-14:
while trying "convex_hull_xy", the "cross_vectors_xy" needs 2 inputs, while 3 is given.

brgcode, 2018-06-18:
solved with hotfix 0.1.0-#96

brgcode, 2018-06-18:
something like this should work now...

```python
from compas.geometry import pointcloud_xy
from compas.geometry import convex_hull_xy
from compas.plotters import Plotter
from compas.utilities import pairwise

cloud = pointcloud_xy(50, (0, 100), (0, 100))
hull = convex_hull_xy(cloud)

points = []
for a in cloud:
    points.append({
        'pos'       : a,
        'facecolor' : '#0000ff',
        'radius'    : 0.5
    })

lines = []
for a, b in pairwise(hull + hull[:1]):
    lines.append({
        'start' : a,
        'end'   : b,
        'color' : '#ff0000',
        'width' : 2.0
    })

plotter = Plotter()

plotter.draw_points(points)
plotter.draw_lines(lines)

plotter.show()

```

closed 2018-06-18 by brgcode