compas-dev/compas #101: Breadth-first tree

augustogandia, 2018-07-04:
Analogous to the `depth_first_tree` function in `compas.topology`, it would be nice to have a `breadth_first_tree` returning predecessors and paths.

One possible implementation would be:

```python

def breadth_first_tree(adjacency, root):
    tovisit  = deque([root])
    visited  = set([root])
    ordering = [root]
    predecessors = {}
    paths = []
    while tovisit: 
        node = tovisit.popleft()
        is_leaf = len(set(adjacency[node]) - visited) == 0
        for nbr in adjacency[node]:
            if nbr not in visited:
                predecessors[nbr]=node
                tovisit.append(nbr)
                visited.add(nbr)
                ordering.append(nbr)
        else:
            path = [node]
            while path[-1] in predecessors:
                path.append(predecessors[path[-1]])
            paths.append(reversed(path))
    return ordering, predecessors, paths
```

brgcode, 2019-02-07:
pull request?

brgcode, 2019-07-07:
@augustogandia no PR?

closed 2020-04-14 by brgcode