compas-dev/compas #103: Making compas available for IronPython/Rhino/Grasshopper

gonzalocasas, 2018-07-11:
Right now the recommended (and documented) way of making `compas` available in an IronPython/Rhino/Grasshopper context is to manually add the paths using the `EditPythonScript` of Rhino.

This is pretty inconvenient, so we should find a better alternative.

I see at least two possible options:

1) IronPython (outside Rhino) uses the `IRONPYTHONPATH` env variable the same way CPython uses `PYTHONPATH`. But this is **not** honored by Rhino. One option is to make sure `compas` gets added there, and then have a bit of bootstrapping code to read that and setup `sys.path` accordingly.

    <details>
    <summary>Something like this</summary>

        import os
        import sys

        iron_python_paths = os.environ['IRONPYTHONPATH'].split(';')
        for p in iron_python_paths:
            if p not in sys.path:
                sys.path.append(p)

    </details>

2) Rhino keeps the search paths in an XML file, it would be possible to use the installer to find that, and update to the installation path. We did something similar for our [python template cookiecutter](https://github.com/gramaziokohler/python_template), the [post-hook of the template](https://github.com/gramaziokohler/python_template/blob/master/hooks/post_gen_project.py#L27) goes and updates the file to match the project folder.

Ideally, the solution we go for should support virtual environments. :)

gonzalocasas, 2018-07-11:
There's also the approach used by `gh-python-remote` of using `pip`'s api: https://github.com/Digital-Structures/ghpythonremote/blob/master/ghpythonremote/_configure_ironpython_installation.py

brgcode, 2018-07-23:
something like this?

https://gist.github.com/brgcode/c46045de077be438e80c817c2cdc5cf2

gonzalocasas, 2018-07-23:
Yep. Some of these things were already resolved in the code I refered to before ([see here](https://github.com/gramaziokohler/python_template/blob/master/hooks/post_gen_project.py)), like encoding differences between Python 2.x and 3.x and supporting both Rhino 5 and 6, but basically, yes, this would be one approach.

brgcode, 2018-07-23:
should this be integrated in core compas, such that it can be run from the command line?

gonzalocasas, 2018-07-23:
I think so. We could have it as a task via PyInvoke (i.e. `tasks.py`)