compas-dev/compas #106: Problem in mesh.is_manifold because of mesh.vertex_neighbours

robin-oval, 2018-07-20:
In compas.datastructures.mesh.mesh.py, mesh.is_manifold does not detect non-manifold meshes like |><| with two tri faces sharing only one vertex, which results in two open fans for this vertex.

It comes from the use of mesh.vertex_neighbours (and deeper, the use of mesh.face_vertex_descendant) that is not valid for non-manifold meshes.

I would suggest using  only mesh.halfedge to detect if a mesh is manifold or not. The following can be added to detect vertices with multiple fans:

    for key in mesh.vertices():
        ...
        # additional check for multiple fans
        nbr_faces = list(mesh.halfedge[key].values())
        if nbr_faces.count(None) > 1:
            return False

brgcode, 2018-07-23:
hi robin,

can you make a suggestion for the entire method? and add some data for testing?

thanks!

robin-oval, 2018-07-23:
Of course!

    def is_manifold_corrected(self):
        """Verify that the mesh is manifold.

        A mesh is manifold if the fllowing conditions are fulfilled:

        * Each edge is incident to only one or two faces.
        * The faces incident to a vertex form a closed or an open fan.

        Returns
        -------
        bool
            True, if the mesh is manifold.
            False, otherwise.

        """
        if not self.vertex:
            return False

        for key in self.vertices():

            # new initial check without resorting to vertex_neighbours:
            # if multiple fans around vertex (i.e. at least two 'None' faces around it, then not manifold
            # as for |><|
            nbr_faces = list(mesh.halfedge[key].values())
            if nbr_faces.count(None) > 1:
                return False

            nbrs = self.vertex_neighbours(key, ordered=True)

            if not nbrs:
                return False

            if self.halfedge[nbrs[0]][key] is None:
                # check corrected and taken into account above
                #for nbr in nbrs[1:-1]:
                #    if self.halfedge[key][nbr] is None:
                #        return False

                if self.halfedge[key][nbrs[-1]] is not None:
                    return False
            else:
                for nbr in nbrs[1:]:
                    if self.halfedge[key][nbr] is None:
                        return False

        return True

    # non-manifold example

    vertices = [
                [0,0,0],
                [1,1,0],
                [1,-1,0],
                [-1,-1,0],
                [-1,1,0]
                ]
    faces = [
            [0,2,1],
            [0,4,3]
            ]

    mesh = Mesh.from_vertices_and_faces(vertices, faces)

    print('(OLD) is mesh manifold?')
    print(mesh.is_manifold())

    print('(NEW) is mesh manifold?')
    print(mesh.is_manifold_corrected())

    plotter = MeshPlotter(mesh, figsize=(10, 7))
    plotter.defaults['vertex.fontsize'] = 8

    plotter.draw_vertices(text={key: str(key) for key in mesh.vertices()}, radius=0.05)
    plotter.draw_faces(text={fkey: str(fkey) for fkey in mesh.faces()})
    plotter.draw_edges()
    plotter.show()

    # manifold example

    vertices_2 = [
                [0,0,0],
                [1,0,0],
                [1,1,0],
                [0,1,0]
                ]

    faces_2 = [
                [0,1,2,3]
                ]

    mesh = Mesh.from_vertices_and_faces(vertices_2, faces_2)

    print('(OLD) is mesh 2 manifold?')
    print(mesh.is_manifold())

    print('(NEW) is mesh 2 manifold?')
    print(mesh.is_manifold_corrected())

    plotter = MeshPlotter(mesh, figsize=(10, 7))
    plotter.defaults['vertex.fontsize'] = 8

    plotter.draw_vertices(text={key: str(key) for key in mesh.vertices()}, radius=0.05)
    plotter.draw_faces(text={fkey: str(fkey) for fkey in mesh.faces()})
    plotter.draw_edges()
    plotter.show()

closed 2018-08-20 by brgcode