compas-dev/compas #119: Suggestion - Change unload_modules() location

Achillx, 2018-07-31:
When working in Rhino for Mac and Atom, imported modules do not update if you make changes to them. Quitting and starting again Rhino and Atom fixes the issue, but it is annoying!

Using `unload_modules()` from `compas_ghpython` fixes the issue, but unfortunately you cannot import it because it raises an exception:
`Message: 'LightException' object has no attribute 'components'`

Since it does not solve a grasshopper only problem, but a general Rhino issue, it would be more useful and logical to have it in `compas_rhino`. You anyway have to import it if you are working with Rhino!

gonzalocasas, 2018-07-31:
/cc @romanarust

romanarust, 2018-07-31:
Hi there,

Seems that `ghpythonlib` is only available for windows, as reported here:
https://discourse.mcneel.com/t/runtime-error-missingmemberexception-lightexception-object-has-no-attribute-components/41792.
It was only used in some Transform function, so I removed it and replaced it in the last commit.

Yes, I agree, `unload_modules()` should ideally be moved to compas_rhino.
@brgcode:  should I move it into utilities/misc?

brgcode, 2018-08-06:
yup, makes sense...

gonzalocasas, 2018-08-09:
done on develop: https://github.com/compas-dev/compas/commit/f77400165ceba86621c37550496468f728fc1ce1

commit f7740016 (2018-08-08): fix #119 moved unload_modules into compas_rhino and keep it imported in compas_ghpython
closed 2018-08-09 by gonzalocasas