compas-dev/compas #123: bbox.py bug

tkmmark, 2018-08-09:
**change**
x, y = zip(*points)[:2]
**to**
points=[point[:2] for point in points]
x, y = zip(*points)

brgcode, 2018-08-18:
because the first version is less efficient? it basically does 

```python
result = zip(*points)
x, y = result[:2]
```

tkmmark, 2018-08-18:
TypeError: 'zip' object is not subscriptable

romanarust, 2018-08-18:
`x, y = list(zip(*points))[:2]`

Because zip returns zip object in python 3, needs to be converted to a list.

tkmmark, 2018-08-18:
Yes, that is the point.

gonzalocasas, 2018-08-18:
For these cases, it'd be ideal to add a unit test to make sure we don't introduce regressions down the road.

I added one on `develop` branch at https://github.com/compas-dev/compas/commit/ed598f7aa4b7ea06663ae0dee5706eb39b731b37 and fixed the implementation using `islice`. 

Wrapping it in a `list` also works of course, but `islice` is a tad faster. I just timed it with `timeit` on 3 million iterations, and it averages at `islice=7.5sec` vs `list=8.1sec`.

tkmmark, 2018-08-18:
unit test is a great idea!

tkmmark, 2018-08-18:
Unit testing sounds like a great idea!

brgcode, 2018-08-19:
okay, solved!

will keep this to have a central location to report similar issues...

commit ed598f7a (2018-08-18): fix #123: use islice instead of subscripting zip so that it works on 2.x and 3.x
closed 2018-08-19 by brgcode