compas-dev/compas #124: split up compas.robots.model into one class per module

brgcode, 2018-08-09:
this is the convention used throughout the rest of the framework....

gonzalocasas, 2018-08-09:
@brgcode one class per module, for this size of classes (many of them are 5 to 10 lines) seems overkill and not very pythonic. How about split into a `model` package and 3 modules inside of it: `robot`, `link` and `joint`, each containing the main class (`Robot`, `Link`, and `Joint` respectively) and also the classes that go tightly together, e.g. `Limit`, `Dynamics`, `Calibration`, etc go together with `Joint`.

Sounds good?

gonzalocasas, 2018-08-09:
this would also match the split on the documentation; https://compas-dev.github.io/main/reference/compas.robots.html

brgcode, 2018-08-09:
sounds good