compas-dev/compas #125: properties v methods

brgcode, 2018-08-09:
hi,

in most xforms and in frame, some properties are implemented as descriptors and others as methods. the reason seems to be that some require parameters to be passed and others not, which makes sense.

however, the user cannot know this and therefore this is confusing.

for example, ``basis_vectors`` is implemented as a read-only descriptor, whereas ``euler_angles`` is a method because it accepts parameters such as ``static=True`` and ``axes='xyz'``.

any suggestions for handling this more consistently?

thanks!

brgcode, 2018-08-09:
one possible solution, would be to use the ``property`` decorator in both cases, and document that the underlying getter function can still be accessed directly. for example:

```python
class Rotation(Transformation):
    ...
    @property
    def euler_angles(self, static=True, axes='xyz'):
        # do stuff
    ....

R = Rotation(...)
a1 = R.euler_angles
a2 = R.get_euler_angles(static=False)

```

romanarust, 2018-08-14:
Yes, euler angles are quite tricky, because different programs or robots use different conventions. 
I would agree with: descriptor R.euler_angles (with default args) and method R.get_euler_angles(), however no getter functions for the rest.

brgcode, 2018-08-18:
well, just to be clear, i just figured out that the getter function is always there when using the property decorator. we just need to make the user aware of it...

brgcode, 2018-08-19:
i just realised that my earlier suggestion was not correct. it should be like this:

```python
class Rotation(Transformation):
    ...
    @property
    def euler_angles(self, static=True, axes='xyz'):
        # do stuff
    ....

R = Rotation(...)
a1 = R.euler_angles
a2 = Rotation.euler_angles.fget(R, static=False)
```

brgcode, 2018-08-19:
but doesn't seem very transparent. i will leave it as a method for now, but it is a general inconsistency that we need to solve...

brgcode, 2019-02-07:
is this still relevant?

gonzalocasas, 2019-02-07:
not as an issue. we could migrate it to the forum in case someone feels like presenting really strong arguments for one or the other, but as it is, it's not an actionable item imo.

closed 2020-01-12 by gonzalocasas