compas-dev/compas #127: spelling consistencies

tkmmark, 2018-08-12:
Example:

**(1)**
def **normalize**_vector(vector):
    """**Normalise** a given vector.
(notice that the method name and the description use different English spelling)

**(2)**
some methods use **-or**: eg: color
other methods use **-our**: eg: neighbour

-recommend switching entirely to British English

brgcode, 2018-08-15:
will do

brgcode, 2018-08-15:
but not now, because it was not as easy as i thought :)

tkmmark, 2018-08-15:
no worries, not urgent at all.

gonzalocasas, 2018-08-17:
> -recommend switching entirely to British English

Maybe I'm biased, but I would find it awkward for an API to be in british English; I am not sure if there's a way to determine which is more common. Checking Rhino and Blender APIs quickly, it does feel like they use US English. The .NET Framework certainly uses US English. Python std lib uses mostly US English I believe, although I'm not 100% certain, but a search on the docs doesn't yield many british terms.

brgcode, 2018-08-17:
initially i thought this was a good idea, but after trying to implement it i realised it indeed creates a number of counter-intuitive situations. perhaps we just need to go for consistent US english?

tkmmark, 2018-08-17:
[image]

gonzalocasas, 2018-08-17:
A map is not really an accurate indicator of the preferred style of an API ;)

I just checked, and [PEP-0008](https://www.python.org/dev/peps/pep-0008/#comments) actually indicates US English because it recommends a famous US-English writing style guide:

 > When writing English, follow [Strunk and White](https://en.wikipedia.org/wiki/The_Elements_of_Style)

The remark is for comments, but since consistency is the keyword here, it goes without saying that code should adhere to the same convention.

brgcode, 2018-08-18:
i think that is a convincing argument, because we try to stick to the default python conventions in every other aspect of the framework as well.

by the way:
https://en.wikisource.org/wiki/The_Elements_of_Style
https://en.wikipedia.org/wiki/The_Elements_of_Programming_Style

closed 2018-08-19 by brgcode

tkmmark, 2018-08-19:
Yes, I read that too.
But is the issue closed?
Has everything switched to US English?