compas-dev/compas #130: suggested function 'mirror_line_plane'

tkmmark, 2018-08-15:
def mirror_line_plane(line, plane):
    pA = closest_point_on_plane(line[0], plane)
    pB = closest_point_on_plane(line[1], plane)
    vecA = subtract_vectors(pA, line[0])
    vecB = subtract_vectors(pB, line[1])
    new_line = [add_vectors(pA, vecA),add_vectors(pB, vecB)]
    return new_line

brgcode, 2018-08-19:
do you know how to submit a pull request?
or would you like me to add the function for you?

tkmmark, 2018-08-19:
I chatted with the post-doc and they mentioned I should write an write an issue about every discrete chunk of code I might want to add or change, and then wait for your green light before submitting?

gonzalocasas, 2018-08-19:
Since we share quite a bit of the dev workflow tooling now (pyinvoke, etc), you could take this [CONTRIBUTING.rst](https://github.com/gramaziokohler/roslibpy/blob/master/CONTRIBUTING.rst), update the links to `compas-dev`'s repo and be done with it. 😉

tkmmark, 2018-08-20:
Tom, yes, please provide some instructions...
Question:
(1) Would we have to do a separate pull request for every single little chunk we edit/add? I can see the creating separate feature branch for a big set of new anticipated features that can be neatly packaged and singled out in a ... but generally, I try to stay and work within the dev branch to stay up-to-date with the changes you are making to the code. 
(2) Also, there are a lot of suggested changes or new features that are much smaller in scale, such as this one, and other typos or errors we have identified and correct.  Please provide instructions appropriate for these type of changes too that hopefully will not require too much additional interruption.  In my opinion, it would be nice if students can be trusted, and be given access to push directly to the dev branch for those type of issues... as is

gonzalocasas, 2018-08-20:
In my opinion, a workflow based on pull requests is not about lack of trust at all. Even having push rights, I prefer sending pull requests for almost everything because:

 - It helps share information, in particular in a team that is not always physically co-located. People won't read commit messages to follow what has changed, but they will notice PRs and might have more info, or just become aware of a new feature that's being added.
 - It works nicely with continuous integration tooling. Everytime a branch or a PR is created, [travis-ci](https://travis-ci.com) runs all unit tests on all configured python version combinations, re-builds documentation and potentially, prepares a release to be ready to deploy. ([current build status for compas](https://travis-ci.com/compas-dev/compas)).
 - It's does not add overhead. Creating a branch and starting a PR from that is basically a couple of clicks.

There's also the admittedly more debatable argument that this is the standard modus operandi on most open source projects. 😉

tkmmark, 2018-08-20:
For example, I just re-pulled from Compas-dev and had to redo corrections I had previously made, such as changing 'vis=GA_Plotter()' to 'vis=GAPlotter()' in **def visualize_evolution:**

Even though I had already made an issue about it weeks ago, 
https://github.com/compas-dev/compas/issues/114
However, that hasn't been incorporated yet.  

In my opinion it would be easier if we could just push directly to the repo, which we currently cannot because we do not have the proper authorisation.  I think it would be good if we could be trusted to make these type of small contributions; it will save everyone time.  I'd hate to bug you for these type of things.

gonzalocasas, 2018-08-20:
But you didn't send a pull request for that, you sent an issue. For a maintainer, fixing an issue is significantly more effort that approving and merging a pull request. If you had the fix already, instead of writing it as text, you could put it in a branch, open a pull request, and then continue working on your stuff. Locally, you would regularly merge from origin, even if the PR is not yet merged, and nothing would be lost.

tkmmark, 2018-08-20:
I look forward to seeing instructions... and maybe a quick tutorial on how to do this efficiently...
I certainly would like to skip the writing altogether.
Again I should preface my comments by saying I am not an advanced user of GitHub or the common protocols and work practices.  This is the second time I have worked with it--the first time begin when I was working in the development team in an engineering company and we are all pushing directly into the repo, even though we also make use of branching and pull requests when we are merging with master, etc.

andrewtliew, 2018-12-01:
In general pull requests needs to be the way to go, otherwise there will be too many people with write access and it will become a bit of a free-for-all. Write access should be for those committing on a very frequent basis.

Admittedly though, many of us need to check and accept pull requests more often, something we need to be more fluent at.

andrewtliew, 2018-12-03:
A good time to introduce the new contributions page:

https://compas-dev.github.io/main/contributions.html

closed 2018-12-03 by andrewtliew