compas-dev/compas #133: toctree seems incorrect on class template

gonzalocasas, 2018-08-19:
The `toctree` settings assigned in the [`class.rst` template](https://github.com/compas-dev/compas/blob/develop/docs/_templates/autosummary/class.rst) seems to be duplicated with the ones set throughout the docs. The template sets it as `:toctree: generated/` but the same setting is also assigned on the docstrings, so it ends up being `generated/generated/`, as visible on travis' output: 

[image]

On our package, that also uses the same custom template, I fixed it changing the `class.rst` toctree to `:toctree: .`, without any apparent side effect.

brgcode, 2018-08-19:
this is just because it uses a relative path. each file automatically generated gets added to a *generated* folder relative to the parent. this is why the nesting occurs. i prefer to have generated files separate from non-generated ones. however, have set an absolute path to prevent the nesting.

it is running while i am writing this...

brgcode, 2018-08-19:
build failed because travis does not allow access to the folder with an absolute root path. removed the directory argument...

closed 2018-08-19 by brgcode