compas-dev/compas #135: list conversions involved in operations on XForms

brgcode, 2018-08-19:
we need a more elegant and robust solution for this.
for example:

```python
def transpose_matrix(M):
    """Transpose a matrix.

    Parameters
    ----------
    M : sequence of sequence of float
        The matrix to be transposed.

    Returns
    -------
    list of list of float
        The result matrix.

    """
    return list(list(item) for item in zip(* list(M)))
```

romanarust, 2018-08-20:
`list(map(list, zip(*M)))`

brgcode, 2018-08-20:
yes that is already better, but that is not what i meant.

however, i just realised that even if the matrix would be a list of lists we would only be able to get rid of the inner most conversion that is apparently required for compatibility with ironpython:

```python
# ironpython
list(map(list, zip(*M)))         # list of list of float
list(map(list, zip(*list(M))))  # support for XForm objects

# cpython
list(map(list, zip(*M)))         # in both cases
```

the other conversions are because zip returns a sequence of tuples and map returns a map object in py3. so the difference is minor...

closed 2018-08-20 by brgcode