compas-dev/compas #138: Add artists to compas_ghpython

gonzalocasas, 2018-08-19:
The new convention for drawing on CAD with COMPAS is to use the `Artists` interface (e.g. `MeshArtist`, `NetworkArtist`). `compas_rhino` and `compas_blender` already support that, and it would make sense to bring them to `compas_ghpython`.

However there are a couple of questions that need to be defined upfront:
  * Is the `Artists` interface already documented? Or the concept of them? For the longest time, I didn't even realize the term **artists** was a COMPAS-specific concept.
  * Correct me if I'm wrong, but `MeshArtists` do not actually create meshes in the current implementations, instead, they draw vertices, faces, edges, and so on, all as separate objects. Is that really desirable? (both in terms of user experience since the created objects are not grouped, as well as in terms of performance).
  * Since no objects are really created in ghpython, all the `clear*` methods would be superfluous. Is that correct?

This issue stems from the discussion raised on https://github.com/compas-dev/compas/pull/134.

gonzalocasas, 2018-08-20:
A started a branch ([ghpython_artists](https://github.com/compas-dev/compas/tree/feature/ghpython_artists)) with a very simplistic implementation of `MeshArtist` for `ghpython`. I did find a useful use for the `clear*` methods by keeping objects to draw in a local buffer (dict) so that it roughly matches the stateful nature of other artists' implementations that base on document state.

In any case, the branch is far from final, is only a sketch of how this could be implemented, and to serve as basis for discussion.

brgcode, 2018-08-20:
`MeshArtist` draws all faces separately or as one big mesh depending on the flag `join_faces`. the reason for this is that in the context of some algorithms it makes sense to be able to color the faces separately, and to be able to select them individually. the vertices and edges are indeed not grouped into one object, because rhino does not allow for this. however, many algorithms require vertices and edges to be *selectable* such that there attributes can be modified, for example for form finding...

gonzalocasas, 2018-09-18:
I sketched one option at this topic on https://github.com/compas-dev/compas/tree/feature/ghpython_artists

gonzalocasas, 2018-11-06:
Already merged (https://github.com/compas-dev/compas/pull/187)

closed 2018-11-06 by gonzalocasas