compas-dev/compas #143: Revert `requests` dependency

gonzalocasas, 2018-08-27:
As much as I like `requests`,  I think we need to revert commit https://github.com/compas-dev/compas/commit/26c0f5e171f21df419780257c3483cbc865f130b which added a dependency on it (which btw would be missing on the `requirements.txt`)

The main reason to request the revert is CAD integration. In particular, Rhino, that runs on IronPython, makes it really hard to install additional dependencies and adding this one for a relatively basic feature feels like an avoidable issue.

brgcode, 2018-08-28:
agreed. i will add a utility function that replaces all the individual calls to ``requests`` and ``urllib`` so that it is easier to update in the future...

brgcode, 2018-08-28:
done

closed 2018-08-28 by brgcode