compas-dev/compas #147: Reconsider file metadata convention

gonzalocasas, 2018-08-27:
Currently, there's the convention of starting every single file in the framework with a metadata block like this one:
```python
__author__    = ['xxx', ]
__copyright__ = 'Copyright 201x - xxx'
__license__   = 'MIT License'
__email__     = 'xx@xx.xx'
````

The info is however highly redundant. Git itself holds proper history about how did what and when, the license field is just redundant for no reason, and the copyright field is mostly outdated on half the files by now, and would require manual caring (or scripted yearly updates at best) to be kept in sync.

I would suggest getting rid of them globally, and relying only on top level package metadata fields, and git history for understanding contribution.

brgcode, 2018-08-28:
alright, making the change now.
added a CONTRIBUTORS file where people can be listed or list themselves.
can you add the GKR people there?

gonzalocasas, 2018-08-29:
done!

closed 2018-08-29 by gonzalocasas