compas-dev/compas #150: PySide dependency not satisfiable on Py2.7+Windows

gonzalocasas, 2018-08-28:
It seems that the new `pyside2` dependency is not satisfiable on Windows with Python 2.7. Python 3.x works but on 2.7, `pip` complains with:

```
  Could not find a version that satisfies the requirement PySide2 (from COMPAS==0.3.0->-r requirements-dev.txt (line 10)) (from versions: )
No matching distribution found for PySide2 (from COMPAS==0.3.0->-r requirements-dev.txt (line 10))
```

gonzalocasas, 2018-08-28:
Does it have to be a hard-required dependency (i.e. specified in the `requirements.txt`) or could we move it to extras?

brgcode, 2018-08-28:
we could move it to extras.
in general, though, could we not make a separate requirements file for 2.7 and 3.6?

gonzalocasas, 2018-08-28:
While it would be possible to create two different files, it will probably lead to more complexity. One way to specify this would be at the conda recipe using [preprocessing selectors](https://conda.io/docs/user-guide/tasks/build-packages/define-metadata.html#preprocessing-selectors).

So, I would suggest to:
 - keep the `requirements.txt` simple, only strictly mandatory dependencies
 - everything else goes to extra
 - on the `conda` recipe, we install as much as possible, including some extras, but there we filter with the preprocessing selectors, e.g. `# [py2k and win]`

brgcode, 2018-08-28:
how do you propose we format the ``extras_require``?
the sample project at https://github.com/pypa/sampleproject/blob/master/setup.py suggests the extras should described in a dict which basically provides *conditional* extras...

gonzalocasas, 2018-08-29:
perhaps splitting them into "feature categories", e.g. `dict(viewers=[], optimizers=[], robotics=[])`?

brgcode, 2018-08-31:
solved in ``91cd359``

closed 2018-08-31 by brgcode