compas-dev/compas #152: Transform Matrices Locally

augustogandia, 2018-08-28:
A common problem when using transformations for expressing for example the propagation of tolerances in a structures is that you concatenate transformations by multiplying them. With the existing functionality you can do it but it is also interestiong to locally transform a frame, meaning that you can translate a frame on its own coordinate system. Here three functions for each axis (X,Y,Z) and one for all combined together.

def translate_frame_own_x(transformation, distance):
    x_transform = [[1, 0, 0, (transformation[0, 0] * distance)],
                   [0, 1, 0, (transformation[1, 0] * distance)],
                   [0, 0, 1, (transformation[2, 0] * distance)],
                   [0, 0, 0, 1]]
    return Translation.from_matrix(x_transform)
    
def translate_frame_own_y(transformation,  distance):
    y_transform = [[1, 0, 0, (transformation[0, 1]*distance)],
                   [0, 1, 0, (transformation[1, 1]*distance)],
                   [0, 0, 1, (transformation[2, 1]*distance)],
                   [0, 0, 0, 1]]
    return Translation.from_matrix(y_transform)
    
def translate_frame_own_z(transformation,  distance):
    z_transform = [[1, 0, 0, (transformation[0, 2]*distance)],
                   [0, 1, 0, (transformation[1, 2]*distance)],
                   [0, 0, 1, (transformation[2, 2]*distance)],
                   [0, 0, 0, 0]]
    return Translation.from_matrix(z_transform)
    
def translate_frame_own_xyz(transformation,x_dist, y_dist, z_dist):
    x_transform = [[1, 0, 0, (transformation[0, 0] * x_dist)],
                   [0, 1, 0, (transformation[1, 0] * x_dist)],
                   [0, 0, 1, (transformation[2, 0] * x_dist)],
                   [0, 0, 0, 1]]
    y_transform = [[1, 0, 0, (transformation[0, 1]*y_dist)],
                   [0, 1, 0, (transformation[1, 1]*y_dist)],
                   [0, 0, 1, (transformation[2, 1]*y_dist)],
                   [0, 0, 0, 1]]
    z_transform = [[1, 0, 0, (transformation[0, 2]*z_dist)],
                   [0, 1, 0, (transformation[1, 2]*z_dist)],
                   [0, 0, 1, (transformation[2, 2]*z_dist)],
                   [0, 0, 0, 0]]
    return Translation.from_matrix(z_transform)*Translation.from_matrix(y_transform)*Translation.from_matrix(x_transform)

brgcode, 2018-08-28:
can you submit this through a pull request?

andrewtliew, 2018-12-03:
Hi @augustogandia many thanks for the suggestions, you can find contribution guidelines here https://compas-dev.github.io/main/contributions.html

closed 2018-12-03 by andrewtliew