compas-dev/compas #163: conda prompt

brgcode, 2018-09-01:
the conda docs recommend using the conda prompt on Windows. we seem to never mention this. is that because it doesn't really matter or should we make a note of it as well?

gonzalocasas, 2018-09-01:
@brgcode I don't think we need to mention it. Any terminal will work, the only thing "special" about the conda prompt is that it activates the base environment.

What would make sense to do (but not for 0.3.1) is to increasingly suggest people should use a conda environments to isolate the install as best practice towards reproducible research. Since 0.3.1, with the rhino installer and the changes we've done, this will already be possible, but it's not mentioned in the docs explicitly (except the contributors part, and the Blender install).

brgcode, 2018-09-01:
had just started to add some instructions to the getting started list :)

gonzalocasas, 2018-09-01:
Cool!

`<super_picky_comment>`
I'd suggest rewording the reasons and example to focus on reproducibility, not old code compatibility, so something like:
`</super_picky_comment>`

```rst
One of the reasons to use virtual environments is to isolate dependencies, to 
be able to create a fully reproducible setup, and to be able to work with
different versions of Python and/or different versions and combinations of 
packages.

For example, if you work on a project that requires a specific version of Python or
a specific version of a package that is incompatible with the ones required by another 
project, you can use environments to make sure both projects can run side-by-side
without constantly having to update your entire development setup.

Create an environment using conda

::

     $ conda create -n my-project

Create an environment with a specific version of Python

::

     $ conda create -n my-project python=2.7

Install **COMPAS** (or other packages) for this environment

::

     $ conda install -n my-project COMPAS

```

closed 2018-09-03 by brgcode