compas-dev/compas #176: intersection_segment_segment_xy

salmamzfr, 2018-09-18:
After I updated my COMPAS to the latest version yesterday, the intersection_segment_segment_xy does not function correctly.  For example for the lines ((247.0, 189.0),(247.0, 188.0)) and ((246.0,189),(248.0,189.0)), the function returns None.

Also, sometimes when I change the order of points for one line in the input, the function output changes from an intersection point to None.  For example for ((248.0, 188.0), (247.0, 190.0)) and ((248.0, 188.0), (249.0, 188.0)) returns (248.0, 188.0), but for ((247.0, 190.0),(248.0, 188.0)) and ((248.0, 188.0), (249.0, 188.0)) returns None!

brgcode, 2018-09-28:
thanks for reporting. having a look now...

Rippmann, 2018-09-28:
I couldn't reproduce the eeror with the given points above. But something is indeed not working.

e.g. line pair:

([-18.05105004113592, -20.221692727960868, 0.0], [-28.056658399351893, -2.115065342520583, 0.0], [-14.419906050317167, -18.793313906826114, 0.0], [-13.313163253090753, -10.448679242591494, 1.9865751254080228])

also return None so ->"parallel" which is clearly wrong.

brgcode, 2018-09-28:
that is exactly what we implemented (albeit slightly reformulated).
however, we use a tolerance value instead of checking for `d` being zero.
this always fails when d is negative.
the function seems to work correctly if i compare the absolute value of `d` with the tolerance...

brgcode, 2018-09-28:
should be solved in `3f6fac8`
please check and let me know, so i can close the issue...