compas-dev/compas #178: xml module in files confuses imports

brgcode, 2018-09-23:


gonzalocasas, 2018-09-23:
What does it mean "confuses imports"?

brgcode, 2018-09-26:
happens whenever i run `compas.files.ply.py`

but no worries, will run a few more tests to figure out exactly what is going wrong and propose a fix...

brgcode, 2018-09-26:
```
Traceback (most recent call last):
  File "/Users/vanmelet/Code/compas-dev/compas/src/compas/files/ply.py", line 432, in <module>
    from compas.datastructures import Mesh
  File "/Users/vanmelet/Code/compas-dev/compas/src/compas/datastructures/__init__.py", line 62, in <module>
    from .network import *
  File "/Users/vanmelet/Code/compas-dev/compas/src/compas/datastructures/network/__init__.py", line 1, in <module>
    from .network import *
  File "/Users/vanmelet/Code/compas-dev/compas/src/compas/datastructures/network/network.py", line 12, in <module>
    from compas.files import OBJ
  File "/Users/vanmelet/Code/compas-dev/compas/src/compas/files/__init__.py", line 95, in <module>
    from .urdf import *
  File "/Users/vanmelet/Code/compas-dev/compas/src/compas/files/urdf.py", line 7, in <module>
    from compas.files.xml import XML
  File "/Users/vanmelet/Code/compas-dev/compas/src/compas/files/xml.py", line 5, in <module>
    import xml.etree.ElementTree as ET
  File "/Users/vanmelet/Code/compas-dev/compas/src/compas/files/xml.py", line 5, in <module>
    import xml.etree.ElementTree as ET
ModuleNotFoundError: No module named 'xml.etree'; 'xml' is not a package
```

brgcode, 2018-09-26:
i also realise this was not the most useful bug report :)
wrote in the middle of crunch time at AAG as a reminder that this needed to be looked at and added you so i would not forget...

gonzalocasas, 2018-09-26:
@brgcode ah! I see! Running the file with `-m` works (i.e. `python -m compas.files.ply`, but if you want to run it as standalone files (we would need to rename the `xml.py` module to something else (maybe `_xml.py`) to avoid shadowing the std lib `xml` package (when running with -m, then then `from __future__ import absolute_import` takes precedence and this works.

What do you think? Should we go for renaming to `_xml.py`? Or have a better idea for a name? Or is it ok to run this with the `-m` flag?

brgcode, 2018-09-27:
so far i have always renamed our modules in situations like this. for example `compas.utilities.datetime_.py`. i always used an underscore at the back to indicate that is an extension of existing functionality. underscore at the beginning tends to be used to indicate a protected variable.

gonzalocasas, 2018-09-27:
👍  I'll rename to `xml_.py` then

commit 5a71e2c7 (2018-10-03): fix: #178 Do not shadow stdlib xml package
commit 688f4874 (2018-10-03): Merge pull request #188 from compas-dev/fix/xml

fix: #178 Do not shadow stdlib xml package
closed 2018-10-03 by brgcode