compas-dev/compas #182: intersection_line_line(l1, l2, tol=1e-6):

Rippmann, 2018-09-28:
returns unexpected results.

Rippmann, 2018-09-28:
solved by normalizing the plane vectors:

    n1 = normalize_vector(cross_vectors(ab, n))
    n2 = normalize_vector(cross_vectors(cd, n))

should work now. tested it with:

import rhinoscriptsyntax as rs

from compas_rhino.utilities import get_line_coordinates

from compas.geometry import intersection_line_line
from compas.geometry import intersection_line_line_xy

crv1 = rs.GetObject("Curve1",4)
crv2 = rs.GetObject("Curve2",4)

l1, l2 = get_line_coordinates([crv1,crv2])

pt1, pt2 = intersection_line_line(l1, l2)

if pt1:
    if rs.PointCompare(pt1, pt2):
        rs.AddPoint(pt1)
    else:
        rs.AddLine(pt1, pt2)
else:
    print("parallel")

closed 2018-09-28 by Rippmann

Rippmann, 2018-09-28:
I absolutely agree! I was pragmatic about it because I was basically reporting this bug with the solution in parallel to keep track of things. You were just too quick with your comment :)

gonzalocasas, 2018-09-28:
😉