compas-dev/compas #185: compas_rhino artist stops working and needs script engine resets

Rippmann, 2018-09-28:
After a while working with artist objects I noticed that every once in a while (every ~10th time executing a script in rhino) artist functions such as draw_edges(), draw_vertices() etc. stop working. After a reset of the scrip engine all is fine again. That makes the artist unusable when it comes to tool development (e.g RhinoVAULT). That's of course a pity since the artist offers great and useful features. This problem occurs out of the blue. I did not find any relation to RAM use or repetitive abortion of scripts during debugging. 

I haven't look into this in detail but I could image that this problem originates from purging functions and how objects are deleted before they are redrawn again.

brgcode, 2018-10-01:
can you post a sample script because i cannot reproduce the problem.
for example, if i run the following, everything works as expected...

```python
import compas
import compas_rhino
import compas_tna

from compas.datastructures import Mesh
from compas_rhino.artists import MeshArtist

data = compas_tna.get("aag_08_final_formdiagram.json")
mesh = Mesh.from_json(data)
artist = MeshArtist(mesh, layer="MeshArtist")

for _ in range(100):
    # artist.clear_layer()
    artist.clear()
    
    artist.draw_vertices()
    artist.draw_edges()

    artist.redraw()

```

brgcode, 2018-10-01:
furthermore, the artists just call the corresponding utility functions under the hood: `xdraw_`, `delete_objects`, `clear_layer`, ...

purging has always been the default setting for those, which made sense from a debugging point of view, to not increase the memory being used by Rhino too much while developing.

perhaps i can make this a global setting, which can be turned on or off...

brgcode, 2018-10-01:
you can test purging versus not purging by adding the following to the above script

```python

compas_rhino.PURGE_ON_DELETE = False

```

closed 2019-02-06 by brgcode