compas-dev/compas #190: Add python 3.7 support to the installer/configuration

gonzalocasas, 2018-10-05:
Downloading the latest anaconda now ships with python 3.7 instead of 3.6, but there are several references in compas that only indicate support for up to 3.6 (classiffiers and other stuff), which eventually make the installer fail (without any real reason).

gonzalocasas, 2018-10-05:
/cc @Kathrin3010

brgcode, 2018-10-06:
i can see the classifiers, but what is the other stuff?

gonzalocasas, 2018-10-11:
> i can see the classifiers, but what is the other stuff?

Added the classifier on b0ed22d5a0c333f11fa568496b56b6ea1d71e648, the rest is mostly conda-recipe stuff, but as seen in the [comment on the recipe's repo](https://github.com/conda-forge/compas-feedstock/pull/6#issuecomment-429024394) the real issue is `numba`'s 3.7 package not yet ready.

gonzalocasas, 2018-10-17:
Re-rendering on https://github.com/conda-forge/compas-feedstock/pull/7

brgcode, 2018-10-17:
perhaps numba can become an optional requirement?

gonzalocasas, 2018-10-22:
Fixed, the dependencies that caused this to not install on 3.7 are now fixed. Installing today on Python 3.7 works as expected:
```bash
$ conda install compas
Solving environment: done

==> WARNING: A newer version of conda exists. <==
  current version: 4.5.9
  latest version: 4.5.11

Please update conda by running

    $ conda update -n base conda

## Package Plan ##

  environment location: C:\Users\gcasas\Anaconda2\envs\test-install-37

  added / updated specs:
    - compas

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    sympy-1.3                  |        py37_1000         9.6 MB  conda-forge
    scipy-1.1.0                |   py37h4f6bf74_1        13.5 MB
    numpy-1.15.2               |   py37ha559c80_0          48 KB
    six-1.11.0                 |        py37_1001          21 KB  conda-forge
    numpy-base-1.15.2          |   py37h8128ebf_0         3.9 MB
    tornado-5.1.1              |py37hfa6e2cd_1000         664 KB  conda-forge
    fastcache-1.0.2            |py37hfa6e2cd_1001          30 KB  conda-forge
    numba-0.39.0               |   py37h830ac7b_0         2.5 MB
    matplotlib-3.0.0           |   py37h7b168c4_1         6.7 MB  conda-forge
    certifi-2018.10.15         |        py37_1000         137 KB  conda-forge
    mkl_random-1.0.1           |   py37h77b88f5_1         267 KB
    ca-certificates-2018.10.15 |       ha4d7672_0         170 KB  conda-forge
    sip-4.18.1                 |   py37h6538335_0         269 KB  conda-forge
    freetype-2.9.1             |    he8b6a0d_1004         470 KB  conda-forge
    kiwisolver-1.0.1           |py37he980bc4_1002          60 KB  conda-forge
    llvmlite-0.24.0            |   py37h6538335_0         9.3 MB
    pyqt-5.6.0                 |   py37h764d66f_7         4.5 MB  conda-forge
    mpmath-1.0.0               |             py_1         430 KB  conda-forge
    compas-0.3.4               |        py37_1000         2.0 MB  conda-forge
    pillow-5.2.0               |   py37h08bbbbd_0         648 KB
    mkl_fft-1.0.6              |   py37hdbbee80_0         120 KB
    ------------------------------------------------------------
                                           Total:        55.1 MB

The following NEW packages will be INSTALLED:

    blas:            1.0-mkl
    ca-certificates: 2018.10.15-ha4d7672_0   conda-forge
    compas:          0.3.4-py37_1000         conda-forge
    cycler:          0.10.0-py_1             conda-forge
    decorator:       4.3.0-py_0              conda-forge
    fastcache:       1.0.2-py37hfa6e2cd_1001 conda-forge
    freetype:        2.9.1-he8b6a0d_1004     conda-forge
    icc_rt:          2017.0.4-h97af966_0
    icu:             58.2-vc14_0             conda-forge [vc14]
    intel-openmp:    2019.0-118
    jpeg:            9b-vc14_2               conda-forge [vc14]
    kiwisolver:      1.0.1-py37he980bc4_1002 conda-forge
    libpng:          1.6.34-vc14_0           conda-forge [vc14]
    libtiff:         4.0.9-vc14_0            conda-forge [vc14]
    llvmlite:        0.24.0-py37h6538335_0
    matplotlib:      3.0.0-py37h7b168c4_1    conda-forge
    mkl:             2019.0-118
    mkl_fft:         1.0.6-py37hdbbee80_0
    mkl_random:      1.0.1-py37h77b88f5_1
    mpmath:          1.0.0-py_1              conda-forge
    networkx:        2.2-py_1                conda-forge
    numba:           0.39.0-py37h830ac7b_0
    numpy:           1.15.2-py37ha559c80_0
    numpy-base:      1.15.2-py37h8128ebf_0
    olefile:         0.46-py_0               conda-forge
    openssl:         1.0.2o-vc14_0           conda-forge [vc14]
    pillow:          5.2.0-py37h08bbbbd_0
    pyparsing:       2.2.2-py_0              conda-forge
    pyqt:            5.6.0-py37h764d66f_7    conda-forge
    python-dateutil: 2.7.3-py_0              conda-forge
    qt:              5.6.2-vc14_1            conda-forge [vc14]
    scipy:           1.1.0-py37h4f6bf74_1
    sip:             4.18.1-py37h6538335_0   conda-forge
    six:             1.11.0-py37_1001        conda-forge
    sympy:           1.3-py37_1000           conda-forge
    tk:              8.6.8-vc14_0            conda-forge [vc14]
    tornado:         5.1.1-py37hfa6e2cd_1000 conda-forge
    zlib:            1.2.11-vc14_0           conda-forge [vc14]

The following packages will be UPDATED:

    certifi:         2018.8.24-py37_1001     conda-forge --> 2018.10.15-py37_1000 conda-forge
```

closed 2018-10-22 by gonzalocasas