compas-dev/compas #191: Remove pre-compiled files from repo and add to build script

gonzalocasas, 2018-10-11:
There are four files that are pre-compiled and added to this repo (plus 3 matching DLLs, so one DLL seems to be missing):
 - `test.so`
 - `fd.so`
 - `smoothing.so`
 - `drx_c.so`

All of these have to be excepted from the [manifest check](https://github.com/compas-dev/compas/blob/master/tasks.py#L122) and are generally considered a bad idea.

Additionally, they are currently blocking the possibility of [switching this package to `noarch`](https://github.com/conda-forge/compas-feedstock/pull/6#issuecomment-429024394) in conda.

We should remove them and compile them on install.

andrewtliew, 2018-10-30:
I removed `drx_c.so` (was experimental) as this was only working for some reason on Linux and didn't work on Mac. We think it was related to OpenMP but I happy to remove it till we resolve the problem. There was no `.dll` for this, so the missing one is this one.

gonzalocasas, 2018-10-30:
I don't really know how you folks are building these files. Do you have an external build script to build them?

andrewtliew, 2018-10-30:
So far not, and this is probably one of the problems. A consistent external build script that works reliably would be a good idea. So far my ``.so`` build has been in Sublime with GCC:

``bash
"gcc $file_name -o ${file_base_name} -Wall -march=x86-64 -fopenmp -lgsl -lgslcblas  && ./${file_base_name}"
```

brgcode, 2019-06-16:
i will remove the cpp versions of functions until we have sorted out how to add cross platform support.

closed 2019-06-16 by brgcode