compas-dev/compas #202: center_of_mass_polyhedron

anjalimehrotra, 2018-11-16:
center_of_mass_polyhedron gives incorrect answers when applied to simple geometries.

For a polyhedron with vertices given by m1v and faces given by m1f:
m1v = ([[1.0, 1.0, 2.0],
  [0.0, 0.0, 0.0],
  [1.0, 0.0, 2.0],
  [1.0, 0.0, 0.0],
  [0.0, 1.0, 0.0],
  [1.0, 1.0, 0.0],
  [0.0, 1.0, 1.0],
  [0.0, 0.0, 1.0]])

m1f = ([[4, 6, 7, 1],
 [5, 0, 6, 4],
 [4, 1, 3, 5],
 [3, 2, 0, 5],
 [1, 7, 2, 3],
 [6, 0, 2, 7]])

center_of_mass_polyhedron((m1v,m1f))
Output: (-0.833, -0.75, -1.167)

Actual answer: (0.556, 0.5, 0.778)

gonzalocasas, 2018-11-16:
/cc @brgcode this would be an excellent example to fix and add a test case at the same time

brgcode, 2018-11-19:
alrighty, will do.
seems like some of these functions could do with an update anyway...

brgcode, 2019-02-09:
it seems the code is not wrong, but the algorithm assumes that the cycle directions of the faces are such that the normals point consistently outwards. in your example this is exactly the opposite.

```python

m1v = [
    [1.0, 1.0, 2.0],
    [0.0, 0.0, 0.0],
    [1.0, 0.0, 2.0],
    [1.0, 0.0, 0.0],
    [0.0, 1.0, 0.0],
    [1.0, 1.0, 0.0],
    [0.0, 1.0, 1.0],
    [0.0, 0.0, 1.0]
]

m1f = [
    [4, 6, 7, 1],
    [5, 0, 6, 4],
    [4, 1, 3, 5],
    [3, 2, 0, 5],
    [1, 7, 2, 3],
    [6, 0, 2, 7]
]

m1f = [face[::-1] for face in m1f]

print(centroid_polyhedron((m1v, m1f)))
```
yields `[0.5555555555555556, 0.5, 0.7777777777777777]`

brgcode, 2019-02-10:
will add functions to check for convexity and *positivity* soon, but will consider this issue resolved...

closed 2019-02-10 by brgcode