compas-dev/compas #205: 'Point' object has no attribute 'x'

Achillx, 2018-11-23:
When I'm trying to create a compas Point  I get the following error:

> Message: 'Point' object has no attribute 'x'
>
>Traceback:
>  line 99, in \_\_init\_\_, "/Users/axydis/PROJECTS/python/compas-dev/src/compas/geometry/objects/point.py"

```
from compas.geometry import Point
p = Point(1,2,5)
```

compas 0.3.4
Atom 1.32.2
Mac 10.14
Rhino 5.5.2

Achillx, 2018-11-23:
After restarting both Rhino and Atom I have no error!
I do not know why.

I'll let you know if I figure out what triggers it.

brgcode, 2018-11-23:
please do, because that shouldn't happen :)

gonzalocasas, 2018-11-23:
It could be caused by a script context change. The find_object method is
cached on module-load time and perhaps it's invalid later on if the script
context changes (e.g. if Grasshopper changes it)

brgcode, 2018-11-23:
are you sure? we won't tell anyone if you were ... :)

brgcode, 2018-12-01:
progress?

Achillx, 2018-12-01:
No, I haven’t being able to reproduce it.
But something similar, that also got fixed by restarting Rhino was when rhino artist stoped drawing edges and vertices, was not cleaning the layer, but it was drawing faces!

andrewtliew, 2018-12-03:
Closing, as it appears the error is not being reproduced, good!

closed 2018-12-03 by andrewtliew