compas-dev/compas #206: Delanuay triangulation wrong result

tmsmendez, 2018-11-27:
Some geometries produce wrong results when using the Delanuay triangulation (or mesh from points) functions. For example, this code:

```python
from compas.datastructures import Mesh
from compas.plotters import MeshPlotter

pl = [[2.994817685045075, 10.855606612493078, 0.0],
      [4.185204599300653, 9.527867361977242, 0.0],
      [4.414125159734419, 10.718254276232818, 0.0],
      [5.925000858597267, 9.344730913630228, 0.0],
      [8.900968144236211, 10.809822500406325, 0.0],
      [9.496161601363999, 8.566401008155429, 0.0],
      [7.710581229980631, 7.9254234389408875, 0.0],
      [7.847933566240888, 6.414547740078039, 0.0],
      [3.9104999267801377, 4.9036720412151915, 0.0],
      [5.2909301507195865, 6.342692886748852, 0.0]]

mesh = Mesh.from_points(pl, boundary=None, holes=None)

plotter = MeshPlotter(mesh, figsize=(10, 6))

plotter.draw_vertices(text='key', radius=0.2, picker=10)

plotter.draw_edges()
plotter.draw_faces()
plotter.show()
```

Produces this mesh:
[image]

Missing the edge shown in green.

Rippmann, 2018-12-02:
Yes, that example is confusing. It is indeed the correct delaunay triangulation for the given set of points. The problem starts when you give it the boundary vertices [0, 4, 5, 7, 8, 9, 1, 0] I guess. @tmsmendez  can you not just include the example you showed me earlier last week? 
So, I think the delaunay algorithm is perfectly fine. I would be very surprised if not. But how boundaries and holes are enforced has some flaws that I need to weed out.

brgcode, 2018-12-02:
but `boundary` and `holes` are `None` so the result is correct...

tmsmendez, 2018-12-03:
It is the same example from last week. I indeed forgot to include the boundary, but this also does not return a correct result. So including the boundary:

```python
from compas.datastructures import Mesh
from compas.plotters import MeshPlotter

pl = [[2.994817685045075, 10.855606612493078, 0.0],
      [4.185204599300653, 9.527867361977242, 0.0],
      [4.414125159734419, 10.718254276232818, 0.0],
      [5.925000858597267, 9.344730913630228, 0.0],
      [8.900968144236211, 10.809822500406325, 0.0],
      [9.496161601363999, 8.566401008155429, 0.0],
      [7.710581229980631, 7.9254234389408875, 0.0],
      [7.847933566240888, 6.414547740078039, 0.0],
      [3.9104999267801377, 4.9036720412151915, 0.0],
      [5.2909301507195865, 6.342692886748852, 0.0]]

mesh = Mesh.from_points(pl, boundary=pl, holes=None)

plotter = MeshPlotter(mesh, figsize=(10, 6))

plotter.draw_vertices(text='key', radius=0.2, picker=10)

plotter.draw_edges()
plotter.draw_faces()
plotter.show()
```

I get this result:

[image]

The requested boundary in green.

brgcode, 2018-12-03:
that makes sense.
the `from_points` constructor uses a delaunay triangulation of the given points as a starting point, which (in this case) looks like this:

[image]

to create the boundary it will then check if the centroids of the faces are in the boundary polygon and remove those which are not.

here that means that `018` and `819` will be removed...

brgcode, 2018-12-03:
so as a conclusion, the delaunays are correct. it is just very unclear what the purpose of `Mesh.from_points` is, and perhaps it should be removed...

Rippmann, 2018-12-03:
If one want to have a triangulation that fills up the entire polygon without leaving gaps (that's what I would expect as a user), one does need to do more than just deleting triangles based on the centroid. Either one implements a [CDT](https://link.springer.com/content/pdf/10.1007/BF01553881.pdf) and enforces the edges of the boundary (and holes) to always be part of the triangulation (demands edge-edge intersection checks), or one adds a sanitize function to fill possible "holes" (9,1,0 in this example). Maybe only called if there is a isolated vertex left?!  I will look into that. Let's also discuss this... sounds like an interesting algorithmic problem.

Rippmann, 2018-12-03:
> 
> 
> so as a conclusion, the delaunays are correct. it is just very unclear what the purpose of `Mesh.from_points` is, and perhaps it should be removed...

The delaunay triangulation algorithm (compas python only) is working correctly but not the part of the function handling boundaries (and holes) for the aforementioned issues. So, I think the delaunay function needs some additions/improvements. I think mesh.from_points is a nice convenient function to construct a mesh. We just need to be clear in the docs what it is doing. It basically just wraps around the delauney function. So, maybe it should be called `Mesh.from_points_delaunay`

brgcode, 2018-12-03:
```bash
pip install triangle
```

```python
from compas.datastructures import Mesh
from compas.plotters import MeshPlotter
from triangle import triangulate

points = [
    [2.994817685045075, 10.855606612493078, 0.0],
    [4.185204599300653, 9.527867361977242, 0.0],
    [4.414125159734419, 10.718254276232818, 0.0],
    [5.925000858597267, 9.344730913630228, 0.0],
    [8.900968144236211, 10.809822500406325, 0.0],
    [9.496161601363999, 8.566401008155429, 0.0],
    [7.710581229980631, 7.9254234389408875, 0.0],
    [7.847933566240888, 6.414547740078039, 0.0],
    [3.9104999267801377, 4.9036720412151915, 0.0],
    [5.2909301507195865, 6.342692886748852, 0.0]
]

res = triangulate({
    'vertices': [point[0:2] for point in points],
    'segments': [[0, 1], [1, 2], [2, 3], [3, 4], [4, 5], [5, 6], [6, 7], [7, 8], [8, 9], [9, 0]]}, opts='p')

faces = res['triangles']
mesh = Mesh.from_vertices_and_faces(points, faces)

plotter = MeshPlotter(mesh)
plotter.draw_faces()
plotter.draw_vertices(text='key')
plotter.show()
```

will add `triangle` as a backend in the next release...

brgcode, 2018-12-12:
```python
for fkey in list(mesh.faces()):
    vertices = mesh.face_vertices(fkey)
    v = len(vertices)
    if v > 3:
         data = {'vertices': vertices, 'segments': list(mesh.face_halfedges(fkey))}
         res = triangulate(data, opts='p')
         triangles = res['triangles']
         del mesh.face[fkey]
         for tri in triangles:
             mesh.add_face()
```

brgcode, 2019-02-07:
support for constrained delaunay is now available through triangle

closed 2019-02-07 by brgcode