compas-dev/compas #214: trimmesh_subdivide_loop exception

Achillx, 2018-12-02:
**Describe the bug**
The function returns exception:
```
Message: iteration over non-sequence of type Mesh

Traceback:
  line 581, in trimesh_subdivide_loop, "/Users/axydis/PROJECTS/python/compas-dev/src/compas/topology/subdivision.py"
  line 12, in <module>, "/Users/axydis/PROJECTS/PhD/DDAD/src/frontend/voronoi.py"
```

**To Reproduce**
Steps to reproduce the behavior:
```
from compas.datastructures import Mesh
from compas.topology import trimesh_subdivide_loop
from compas.topology import delaunay_from_points
from compas.topology import voronoi_from_delaunay

from compas.geometry import pointcloud_xy

points = pointcloud_xy(50, (0, 100))
faces = delaunay_from_points(points)
delaunay = Mesh.from_vertices_and_faces(points, faces)

mesh_sub = trimesh_subdivide_loop(delaunay, 1, False)
```

**Expected behavior**
To return a subdivided mesh.

**Desktop (please complete the following information):**
 - OS: MacOS Mojave v.10.14
 - Python version: 2.7.15 :: Anaconda, Inc.
 - Compas commit: 9f94cce0
 - IDE: Atom 1.33.0 x64
 -  Rhino for Mac 5.5.2

brgcode, 2018-12-02:
nice bug report :)
will get right on it...

brgcode, 2018-12-02:
there was indeed something wrong with the scheme: it only worked for closed meshes.
should be fixed since 6e5e0e3

i have tested the following successfully:

```python
from compas.datastructures import Mesh
from compas.topology import trimesh_subdivide_loop
from compas.plotters import MeshPlotter

points = [[0.0, 0.0, 0.0], [1.0, 0.0, 0.0], [0.5, 1.0, 0.0]]
faces = [[0, 1, 2]]
mesh = Mesh.from_vertices_and_faces(points, faces)

subd = trimesh_subdivide_loop(mesh, k=3)

plotter = MeshPlotter(subd, figsize=(10, 7))
plotter.draw_vertices(radius=0.1)
plotter.draw_faces()
plotter.show()
```

```python
from compas.datastructures import Mesh
from compas.geometry import pointcloud_xy
from compas.topology import delaunay_from_points
from compas.topology import trimesh_subdivide_loop
from compas.plotters import MeshPlotter

points = pointcloud_xy(10, (0, 100))
faces = delaunay_from_points(points)
mesh = Mesh.from_vertices_and_faces(points, faces)

subd = trimesh_subdivide_loop(mesh, k=3)

plotter = MeshPlotter(subd, figsize=(10, 7))
plotter.draw_vertices(radius=0.1)
plotter.draw_faces()
plotter.show()
```

brgcode, 2018-12-02:
however, higher numbers of starting points in the pointcloud seem to sometimes generate overlaps...
the algorithm is also clearly not yet optimised for speed :)

Achillx, 2018-12-03:
Yes, that fixed it!

closed 2018-12-03 by brgcode