compas-dev/compas #225: error from compas_ghpython artists

GeneKao, 2019-01-07:
Hi, there's an error when I tried to use the ```artist.draw_faces(join_faces=True)``` in ```compas_ghpython.artists.MeshArtist```.  

The error can be reproduced from the example in [documentation](https://compas-dev.github.io/main/api/generated/compas_ghpython.artists.MeshArtist.html#compas_ghpython.artists.MeshArtist). 

```
Runtime error (TypeErrorException): Parameter must be a Guid or string representing a Guid
Traceback:
  line 206, in JoinMeshes, "C:\Users\user\AppData\Roaming\McNeel\Rhinoceros\5.0\Plug-ins\IronPython (814d908a-e25c-493d-97e9-ee3861957f49)\settings\lib\rhinoscript\mesh.py"
  line 495, in coerceguid, "C:\Users\user\AppData\Roaming\McNeel\Rhinoceros\5.0\Plug-ins\IronPython (814d908a-e25c-493d-97e9-ee3861957f49)\settings\lib\rhinoscript\utility.py"
  line 589, in coercemesh, "C:\Users\user\AppData\Roaming\McNeel\Rhinoceros\5.0\Plug-ins\IronPython (814d908a-e25c-493d-97e9-ee3861957f49)\settings\lib\rhinoscript\utility.py"
  line 66, in draw_faces, "Z:\compas-dev\compas\src\compas_ghpython\artists\mixins\faceartist.py"
  line 8, in script
```

I am using Rhino 5, Grasshopper and IronPython2.7 in windows. 

Thanks!

GeneKao, 2019-01-07:
@gonzalocasas @brgcode I found the problem and fixed the code already. I will create a pull request.

brgcode, 2019-01-07:
cool, i will merge it as soon as it comes in!

closed 2019-01-07 by romanarust

romanarust, 2019-01-07:
Hi thanks for posting, I was working on it too. It actually required also to fix something in ghpython/utilities/drawing.py  (returned breps instead of meshes)

GeneKao, 2019-01-07:
@romanarust yes, I also noticed the ```xdraw_faces``` function was drawing breps instead of mesh. Thanks for the bug fix, I was going to suggest that we can keep the breps method or rename it to something else, so we can extend more function to gh later.

brgcode, 2019-01-07:
so is this now solved?

GeneKao, 2019-01-07:
Yes, solved. It works perfectly now.