compas-dev/compas #227: Documentation of Compas Installation Issues and Common Mistakes during MIT-IAP2019 Workshop

tkmmark, 2019-01-12:
-_Matplotlib_ is unsupported in Winodows on Virtual Machine (tna)
-Importing issues with _AST_ when version other than IronPyton 2.7.5 is used (core)
-Issues when using Rhinoceros 6.0 instead of Rhinoceros 5.0 (core)
-Installation via _Conda_ is not possible in China with or without the use of VPN.  As such, it was necessary to directly clone from source.
-Forgetting to insert the add-on package to system environmental path under the variable 'PYTHONPATH' (core, tna, pattern)
-Using an outdated version of Rhino 5.0 (without the newer service release).  This caused issues because the syntax of _rhinoscriptsyntax_ has changed since the earlier version. (pattern)
-Problems with _XFunc_ in Rhino because COMPAS add-on package has not been added to the system environmental path. (pattern, tna)

brgcode, 2019-01-12:
what do you mean with number 3?

brgcode, 2019-01-12:
5 and 7 are the same?

brgcode, 2019-01-12:
i will add a section with known issues to the docs to address each of these...

brgcode, 2019-01-12:
about nr. 1:

this is a problem i also encountered with the version of `matplotlib` that ships with anaconda on windows. i don't think it has anything to do with the virtual machine.

it goes away when you update `matplotlib`...

```bash
$ conda update matplotlib
```

brgcode, 2019-01-12:
about nr. 2:

this is clearly documented in the docs
https://compas-dev.github.io/main/environments/rhino.html#ironpython-1

it is a known problem of most IronPython versions shipped with Rhino5 on windows and has nothing to do with COMPAS. perhaps we could check this during the installation procedure of `compas_rhino` and warn the user if the version is wrong?

brgcode, 2019-01-12:
about nr. 3:

as i said, please elaborate :)

brgcode, 2019-01-12:
about nr. 4, 5, 7:

i assume you cloned the repos and then (*forgot* to) update the `PYTHONPATH` variable to make the packages available. indeed, `XFunc` or `RPC` will not work.

but you can use `conda` to *install* a package from a local repo (the ones you cloned from github). both `compas` and `compas_tna` can be installed that way. i am not sure about `compas_pattern`, but we can easily add that if necessary. and `compas_tna` can't be installed for Rhino, yet...

brgcode, 2019-01-12:
about nr. 6:

perhaps we could list this on a "known issues" page.

gonzalocasas, 2019-01-12:
> and compas_tna can't be installed for Rhino, yet...

Actually, the rhino install provided on core is intended to be used by
compas_* packages. Check compas_fab for an example that uses that to
install compas_fab an its dependencies on rhino.

brgcode, 2019-01-12:
indeed, i forgot about that. nice!

tkmmark, 2019-01-15:
We are documenting all the issues we saw--some of them are more silly than others.  They are listed here for the sake of completion, and in case they might be items that can be better emphasised in some way in the documentation.

tkmmark, 2019-01-15:
> about nr. 4, 5, 7:
> 
> i assume you cloned the repos and then ("forgot" to) update the `PYTHONPATH` variable to make the packages available. indeed, `XFunc` or `RPC` will not work.
> 
> but you can use `conda` to _install_ packages from local repos (the ones you already cloned from github). both `compas` and `compas_tna` can be installed that way. i am not sure about `compas_pattern`, but we can easily add that if necessary.
> 
> and `compas_tna` can't be installed for Rhino, yet...

Yes, they had forgotten to follow our instructions...without quotes.

tkmmark, 2019-01-15:
> I guess this is a placeholder not to forget, but it'd be easier to handle if we split them.
> 
> For instance, nr 4 is not something we can fix. nr 6 is probably also a "wont fix" imo. Nr 5 and 7, as @brgcode says, sound like the same, and I wonder if it's the fact that it was installed from source that somehow caused it (but if it's really due to forgetting, then it's not a bug)

Will split them up for future issues.
I think Tom is already in the process of creating separate issues for these?

brgcode, 2019-01-16:
> We are documenting all the issues we saw--some of them are more silly than others. They are listed here for the sake of completion, and in case they might be items that can be better emphasised in some way in the documentation.

no need to get defensive.
with the comments we just tried to show that most are not really issues, but just the result of the employed procedures...

tkmmark, 2019-01-17:
> no need to get defensive.
> with the comments we just tried to show that most are not really issues, but just the result of the employed procedures...

Not at all Tom: not being defensive.  Maybe there needs to be a GitHub feature for voice-recorded-issue-posts to convey my mega calmness...
In any case, maybe some of the issues shouldn't be included here at all because they are a bit silly or trivial.  But I posted them anyway because I thought it might be something that was not emphasised enough.
Also, do let us know what you mean by issues with procedure.

tkmmark, 2019-01-17:
> about nr. 3:
> 
> as i said, please elaborate :)

Will look into it again and get back to you.

gonzalocasas, 2019-01-17:
> there needs to be a GitHub feature for voice-recorded-issue-posts to convey my mega calmness...

😉 👍 

One pretty nice way of dealing with issues on some large projects is to actively close issues that cannot be worked on but keeping the space explicitly open for discussion (since one can still comment on closed issues). This makes it easier to keep the backlog of pending tasks groomed. And yep, splitting issues up makes it easier to do that, otherwise closing this one would be too much ;)

brgcode, 2019-01-17:
i made a separate issue of nr. 1, but i don't consider the other problems as bugs.

closed 2019-01-17 by brgcode