compas-dev/compas #229: planarity as dependency

brgcode, 2019-01-14:
`planarity` is `pip`-installable. is there a reason why we are not including it as a dependency?

gonzalocasas, 2019-01-14:
In principle, we could add it, but it has a dependency on `cython` which is not defined in the `pip` package, so it needs to be installed separately. This is what happens when trying to `pip install` it on a clean setup:

```
Collecting planarity
  Downloading https://files.pythonhosted.org/packages/91/d6/33cc8784b375a45505326e28cc01ed987e29a8091aa7c70de0c10aee7704/planarity-0.4.1.tar.gz (164kB)
    100% |████████████████████████████████| 174kB 1.4MB/s
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\gcasas\AppData\Local\Temp\pip-install-dksekeaj\planarity\setup.py", line 8, in <module>
        from Cython.Distutils import build_ext
    ModuleNotFoundError: No module named 'Cython'

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in C:\Users\gcasas\AppData\Local\Temp\pip-install-dksekeaj\planarity\ 
```

After manually triggering the pip install of `cython` it seems to work. We can definitely add to base requirements, but when we do, let's add it to the conda recipe as well, so that the installation cython dependency is handled nicely by conda.

brgcode, 2019-01-14:
so do i add that in the compas-feedstock repo > recipe > meta.yaml?

brgcode, 2019-01-14:
in my fork, and that issue a pr?

brgcode, 2019-01-14:
that == then

gonzalocasas, 2019-01-14:
yup ;)

brgcode, 2019-01-14:
i noticed the version number in the recipe is still `0.3.5`. does this have to be changed manually?
also, the checklist says to bump the build number if the version remains unchanged. the build number is `1000`. technically the version number would remain unchanged, but should i not change it to `0.3.6`? and what do i do with the build number then?

brgcode, 2019-01-14:
should hte hash_value aslo be changed manually? copied from pypi?

gonzalocasas, 2019-01-14:
> version number is the recipe is still `0.3.5`

I think you have an old clone, here it's 0.3.6: https://github.com/conda-forge/compas-feedstock

brgcode, 2019-01-15:
okay, so it seems i did not do this correctly.
on travis the build passes again.

i added
```bash
pip install Cython --install-option="--no-cython-compile"
```
for `Cython` as per the recommendation on their website.

not sure how to proceed with the `conda` thing...

brgcode, 2019-01-15:
`planarity` is now a requirement on not windows

closed 2019-01-15 by brgcode

htlcnn, 2020-04-19:
@brgcode I could not install planarity on Windows. How could I install COMPAS using `python -m pip install COMPAS` and not install `planarity`?

brgcode, 2020-04-19:
you could clone the source, remove planarity from the requirements and install from source.

but it seems there is a problem with your Cython installation and/or the available C/C++ compilers on your system. have you looked at the [Known Issues](https://compas-dev.github.io/main/gettingstarted/knownissues.html)?

i know you don't literally get the same errors as listed there, but the problem sounds remarkably similar. perhaps trying those fixes is also worth a shot.

brgcode, 2020-04-19:
btw, any reason you are using a standard Python install (`C:\Program Files\Python3\python.exe`) and not something like `conda` or `miniconda`?

htlcnn, 2020-04-19:
I cloned the source, remove planarity from requirements.txt. Installation was successful. Does compas run normally without planarity?
I read Known Issues and installed both Cython and VS C++ Build Tools, but somehow Cython could not find Visual Studio as it said.
I've been used to pure python installation from python.org and install packages via pip. conda seems to have many existing packages that I don't need. I tried miniconda but it could not connect to server to download packages, so I switched back to pure python and pip.
Thanks for your help!

brgcode, 2020-04-19:
yes, it will work without planarity. left-over from the days when the functionality of quite a few packages (e.g. compas_ags) was still integrated in the core framework. will remove it soon-ish...