compas-dev/compas #230: Binary STL loading performance issues

gonzalocasas, 2019-01-16:
The performance is loading an STL binary mesh is pretty slow. The slowness seems to come from a few different factors, but mainly the `geometric_key` generation is very slow and the multiple small reads on the file are also slowing down the process (instead of a single longer read for the fixed size per triangle).

I'll send a PR with a proposal later today.

gonzalocasas, 2019-01-16:
Being worked on on #231

brgcode, 2019-01-17:
i merged #231 :)

closed 2019-01-17 by brgcode

gonzalocasas, 2019-01-17:
@brgcode cool, thanks!

regarding the question you posted on the PR about suggesting geo_keys for post-processing, I've been testing a different way to generate geometric keys, it's quite faster than the current but it introduces a breaking change so I need to test quite a bit before sending a PR, but once I do, we can include things like this post-processing as an optional.