compas-dev/compas #236: removing symlink raises NotADirectoryError

brgcode, 2019-01-20:
it seems that removing a symlink is always handled by `os.unlink`, both for links to directories and to files...

gonzalocasas, 2019-01-20:
do you get this using `compas._os.remove_link()`? is it on osx with py3k? could you paste the stack trace of the exception?

brgcode, 2019-01-21:
```
$ python -m compas_rhino.install_plugin ui/RhinoMac/RBE{520ddb34-e56d-4a37-9c58-1da10edd1d62}
```

```
Installing PlugIn RBE{520ddb34-e56d-4a37-9c58-1da10edd1d62} to RhinoMac PythonPlugIns.
Traceback (most recent call last):
  File "/Users/vanmelet/anaconda3/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/Users/vanmelet/anaconda3/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/Users/vanmelet/Code/compas-dev/compas/src/compas_rhino/install_plugin.py", line 81, in <module>
    install_plugin(args.plugin_name)
  File "/Users/vanmelet/Code/compas-dev/compas/src/compas_rhino/install_plugin.py", line 61, in install_plugin
    remove_symlink(destination)
  File "/Users/vanmelet/Code/compas-dev/compas/src/compas/_os.py", line 78, in remove_symlink
    os.rmdir(link)
NotADirectoryError: [Errno 20] Not a directory: '/Users/vanmelet/Library/Application Support/McNeel/Rhinoceros/MacPlugIns/PythonPlugIns/RBE{520ddb34-e56d-4a37-9c58-1da10edd1d62}'
```

closed 2019-02-06 by brgcode