compas-dev/compas #239: Guidelines for creating API documentation?

yijiangh, 2019-01-24:
I am wondering if we have a guideline for writing API documentation? If yes, could anyone points me to the guideline?

Based on my observation, I think we are using [google docstring](https://sphinxcontrib-napoleon.readthedocs.io/en/latest/example_google.html) style. Just want to confirm before I continue with the default [sphinx_docstring](http://www.sphinx-doc.org/en/master/usage/extensions/autodoc.html) style. Other than this, is there anything else that developers need to pay attention to in order to conform to the existing compas doc convention?

romanarust, 2019-01-25:
Great that you intend to contribute!
As far as I know there is no guideline for API documentation yet. 
For docstrings we use the [numpy docstring style](https://sphinxcontrib-napoleon.readthedocs.io/en/latest/example_numpy.html).

yijiangh, 2019-01-25:
@romanarust Thanks for your reply!

Also, I'm wondering if we can have a link to the source code in the API doc, maybe consider using [sphinx.ext.viewcode](http://www.sphinx-doc.org/en/master/usage/extensions/viewcode.html#module-sphinx.ext.viewcode)? Examples can be found in pytorch's API doc [here](https://pytorch.org/docs/stable/torch.html).

This will enable our users to easily navigate to the source who need to know what's going on under the hood when the API doc is not sufficient for understanding. This turns out to be the case for me when I worked with the ROS C++ API. The link to `xxx line of xxx.h/xxx.cpp`, e.g. [see here](http://docs.ros.org/melodic/api/moveit_core/html/classplanning__scene_1_1PlanningScene.html#a92861d0f2e6509860d3d192a4bd8e635), turns out to be very useful for me.

brgcode, 2019-01-31:
> Also, I'm wondering if we can have a link to the source code in the API doc, maybe consider using sphinx.ext.viewcode? Examples can be found in pytorch's API doc here.

those links will be re-added as soon as the docstrings are cleaned up and consistent, which should be soon...

brgcode, 2019-01-31:
> Other than this, is there anything else that developers need to pay attention to in order to conform to the existing compas doc convention?

we have not yet released a "dev guide", but in general we try to stick to the general Python style guides and rules as much as possible...

one thing perhaps, is that we make sure all code is accessible from the "second level". for example,

```python
from compas.datastructures import Mesh
from compas.plotters import MeshPlotter
from compas.geometry import bounding_box
```
regardless of the structure of the package below that level....

brgcode, 2019-01-31:
i would also recommend using the cookiecutter template to create a compas package, because it will set up most of these things for you.

the current version is at https://github.com/BlockResearchGroup/cookiecutter-compas-package

jf---, 2024-04-06:
@tomvanmele its old and irrelevant

closed 2024-04-08 by gonzalocasas