compas-dev/compas #241: Mesh Loading from URDF

arpastrana, 2019-02-07:
**Describe the bug**
In `compas.robots`, `DefaultMeshLoader.can_load_mesh()` promptly returns `True` as it only checks for `url` formatting, and not whether the file exists, if it is an absolute or relative path, or can actually be parsed by the `DefaultMeshLoader.load_mesh()`.  This leads to a premature exception raise that is hard to trace afterwards in line 302 of `robots.resources.basic.py `.
                           
**To Reproduce**
```
from compas.robots.model import RobotModel

robot = RobotModel.from_urdf_file(my_filepath)
robot.load_geometry()
```
**Expected behavior**
The geometry prescribed in the robot's urdf is succesfully loaded, or raises an exception when this is not possible.

At the moment, I am working with a Kuka IIWA urdf file which has `shape.filename` as a relative path (`filename="meshes/link_5.obj"`). Therefore, the meshes cannot be loaded by the currently-available `LocalPackageMeshLoader ` and `DefautMeshLoader`. I wrote a new `LocalFolderMeshLoader` to fix the issue. 

**Desktop:**
 - OS: Windows 10
 - Python version: 3.6
 - Python package manager: Conda

**Aditional Info**
An excerpt of the urdf file is here:

```
<robot name="lbr_iiwa" xmlns:xacro="http://www.ros.org/wiki/xacro">
  <material name="Grey">
    <color rgba="0.2 0.2 0.2 0.5"/>
  </material>
  <material name="Orange">
    <color rgba="1.0 0.423529411765 0.0392156862745 0.5"/>
  </material>
  <material name="Blue">
  <color rgba="0.5 0.7 1.0 0.5"/>
</material>
<material name="beam_color">
    <color rgba="0.72 0.60 0.30 0.75"/>
</material>
  <link name="lbr_iiwa_link_0">
    <inertial>
      <origin rpy="0 0 0" xyz="-0.1 0 0.07"/>
      <mass value="0.0"/>
      <inertia ixx="0.05" ixy="0" ixz="0" iyy="0.06" iyz="0" izz="0.03"/>
    </inertial>
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="meshes/link_0.obj"/>
      </geometry>
      <material name="Grey"/>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="meshes/link_0.stl"/>
      </geometry>
    </collision>
  </link>
  <joint name="lbr_iiwa_joint_1" type="revolute">
    <parent link="lbr_iiwa_link_0"/>
    <child link="lbr_iiwa_link_1"/>
    <origin rpy="0 0 0" xyz="0 0 0.1575"/>
    <axis xyz="0 0 1"/>
    <limit effort="300" lower="-2.96705972839" upper="2.96705972839" velocity="10"/>
    <dynamics damping="0.5"/>
  </joint>
  <link name="lbr_iiwa_link_1">
    <inertial>
      <origin rpy="0 0 0" xyz="0 -0.03 0.12"/>
      <mass value="4"/>
      <inertia ixx="0.1" ixy="0" ixz="0" iyy="0.09" iyz="0" izz="0.02"/>
    </inertial>
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="meshes/link_1.obj"/>
      </geometry>
      <material name="Blue"/>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="meshes/link_1.stl"/>
      </geometry>
    </collision>
  </link>
```

gonzalocasas, 2019-02-07:
@arpastrana thanks for the very detailed report! I'll look into it!

arpastrana, 2019-02-07:
@gonzalocasas Thanks! I have made a pull-request (#242) with some proposed changes for your review.

arpastrana, 2019-02-21:
@gonzalocasas Solved with #242?

gonzalocasas, 2019-02-21:
yep, thx!

closed 2019-02-21 by gonzalocasas