compas-dev/compas #263: Sublime Text 3 does not show plots on Windows

brgcode, 2019-03-07:
**Description**
On windows, Sublime Text 3 now apparently suppresses GUIs. Therefore, COMPAS plotter windows will not appear and the execution of the script never terminates (you have to kill the Python process using your task manager).  

**To Reproduce**
Steps to reproduce the behavior:
1. In Sublime Text 3
2. sample.py (attached)

**Expected behavior**
The plotter window should pop up and display the remeshing process.

**Desktop (please complete the following information):**
 - OS: Windows
 - Python version: any
 - Python package manager: any

brgcode, 2019-03-07:
```python
# samples.py

from compas.datastructures import Mesh
from compas.datastructures import mesh_smooth_area
from compas.datastructures import trimesh_remesh
from compas.plotters import MeshPlotter

vertices = [(0.0, 0.0, 0.0), (10.0, 0.0, 0.0), (6.0, 10.0, 0.0), (0.0, 10.0, 0.0)]
faces = [[0, 1, 2, 3]]

mesh = Mesh.from_vertices_and_faces(vertices, faces)

key = mesh.insert_vertex(0)
fixed = [key]

plotter = MeshPlotter(mesh, figsize=(10, 7))

plotter.draw_edges(width=0.5)

def callback(mesh, k, args):
    print(k)
    plotter.update_edges()
    plotter.update()

trimesh_remesh(
    mesh,
    0.5,
    kmax=200,
    allow_boundary_split=True,
    allow_boundary_swap=True,
    allow_boundary_collapse=True,
    fixed=fixed,
    callback=callback)

mesh_smooth_area(mesh, fixed=mesh.vertices_on_boundary())

mesh_smooth_area(mesh, fixed=mesh.vertices_on_boundary())

plotter.update_edges()
plotter.update(pause=2.0)
plotter.show()
```

brgcode, 2019-03-07:
the problem can be solved by adding a custom builder. unfortunately, that means you can't use *Conda* as your build system...

```
{
	"cmd":
	[
		"start",
		"cmd",
		"/k",
		"%USERPROFILE%\\Anaconda3\\envs\\workshop\\python.exe",
		"-u",
		"$file"
	],
	"shell": true,
	"file_regex": "^[ ]*File \"(...*?)\", line ([0-9]*)",
	"name": "Workshop",
	"selector": "source.python"
}

```

gonzalocasas, 2019-03-14:
PR submitted to the `sublime-text-conda` package ☝️

brgcode, 2019-03-14:
oooh nice!

gonzalocasas, 2019-03-15:
Merged and released on [sublime-text-conda v0.5.2](https://github.com/mandeep/sublime-text-conda/releases/tag/v0.5.2).

closed 2019-03-15 by gonzalocasas